Tinkerforge/warp-charger #2: Failed to deserialize string, using default configuration: EmptyInput

22kW, 2021-03-04:
Habe jede Menge Einträge im Ereignis-Log:

Failed to deserialize string, using default configuration: EmptyInput

[event_log.txt](https://github.com/Tinkerforge/warp-charger/files/6083021/event_log.txt)

rtrbt, 2021-03-04:
Moin,
Lade mal auch den Debug-Report hoch, eventuell sehe ich dann, wo das Problem ist.

22kW, 2021-03-04:
[debug_report.txt.json.txt](https://github.com/Tinkerforge/warp-charger/files/6083051/debug_report.txt.json.txt)

22kW, 2021-03-04:
ist mir aufgefallen beim Testen der warp mit evcc über mqtt

rtrbt, 2021-03-04:
Hast du den Pull-Request https://github.com/andig/evcc/pull/718 verwendet oder dir die Anbindung zwischen EVCC und der Box selbst konfiguriert o.Ä? Ich muss zu geben so tief stecke ich in EVCC nicht drin. Werde aber in den nächsten Tagen mit @andig videochatten, da könnte man das gleich mit klären.

Die Log-Meldung an sich bedeutet, dass irgendeins der MQTT-Topics auf denen die Box subscribt ist, einen leeren Payload bekommen hat, der deshalb ignoriert wird.

22kW, 2021-03-04:
genau, ich teste gerade den Pull-Request andig/evcc#718
Habe es ihm auch schon geschickt.

andig, 2021-03-04:
Bei start/stop schicke leeren payload weil nil nicht möglich ist (gibt paho nicht her). Das scheint mir hier aber deutlich häufiger vorzukommen als dass das die Ursache sein könnte?

rtrbt, 2021-03-04:
> Bei start/stop schicke leeren payload weil nil nicht möglich ist (gibt paho nicht her).

Stand jetzt braucht die Firmware valides JSON (lies: null). Paho nimmt soweit ich das gerade ergoogelt habe interface{} als payload-Typ. Kannst du da nicht "null" also als String reinpacken?

@22kW Teste mal bitte nochmal mit der angehangenen Firmware (musst du erst entpacken, Github mag keine bin-Dateien). Ich habe das Logging etwas umgebaut, damit wir herausfinden, welches Topic das ist.

Nebenbei: Die Fehlermeldungen wurden immer doppelt ausgegeben, das sollte jetzt auch gefixt sein.

[warp_firmware_1_1_1_6040d74c_merged.bin.zip](https://github.com/Tinkerforge/warp-charger/files/6083745/warp_firmware_1_1_1_6040d74c_merged.bin.zip)

andig, 2021-03-04:
> Stand jetzt braucht die Firmware valides JSON (lies: null). Paho nimmt soweit ich das gerade ergoogelt habe interface{} als payload-Typ. Kannst du da nicht "null" also als String reinpacken?

Jetzt verstehe ich endlich was ihr meint.... ja, ist drin!

22kW, 2021-03-04:
so, im Prinzip funktioniert alles, nur ich bekomme in evcc wenn KEIN auto angesteckt ist regelmäßig: 
[lp-1  ] WARN 2021/03/04 17:52:45 charger out of sync
(beim Laden gibts keine WARN)

Das warp log ist jetzt recht leer:
[event_log.txt](https://github.com/Tinkerforge/warp-charger/files/6085487/event_log.txt)

[debug_report.txt](https://github.com/Tinkerforge/warp-charger/files/6085493/debug_report.txt)

andig, 2021-03-04:
Also das passiert immer, wenn EVCC einen Status gesetzt hat, den der Charger nicht hat (out of sync).

@rtrbt das liegt genau an unserer Diskussion bzgl. "enabled state". Code hier https://github.com/andig/evcc/pull/718/files#diff-b914c27fe1f20428b2f0fce78608abe0c2a6075d4a8f274ed74a667e6b26c52bR126 falls Dir noch was auffällt?

rtrbt, 2021-03-05:
@andig Spontan sieht das soweit gut aus. Lass uns da wie gesagt nochmal drüber reden, ich schick dir gleich eine E-Mail.

andig, 2021-03-08:
Refs https://github.com/andig/evcc/pull/718

andig, 2021-03-08:
> so, im Prinzip funktioniert alles, nur ich bekomme in evcc wenn KEIN auto angesteckt ist regelmäßig:

@22kW kannst Du davon bitte _hier_ ein EVCC Logfile anhängen. Gerne mit aktuellem Commit. Und lass uns die komplette Diskussion bitte _hier_ zu Ende führen.

Vielen Dank, Andreas

22kW, 2021-03-08:
O.K. hier ein aktuelles evcc log. 
Beobachtung: Auto eingesteckt, Schütz zieht in kurzer Folge zwei mal an und fällt sofort wieder ab.
Erst dann zieht es ein drittes mal endgültig an und lädt:

[TRACE 2021:03:08 10-34-23.txt](https://github.com/Tinkerforge/warp-charger/files/6100472/TRACE.2021.03.08.10-34-23.txt)

Kurz vorher beim ersten Einstecken hat alles wie erwartet funktioniert (Einstecken, Schütz zieht an, Auto lädt):

[TRACE 2021:03:08 10-32-40.txt](https://github.com/Tinkerforge/warp-charger/files/6100538/TRACE.2021.03.08.10-32-40.txt)

22kW, 2021-03-08:
Muss ich vielleicht in der WARP GUI den Selektor "Ladekontrolle Autostart" auf disable stellen ?

[image]

22kW, 2021-03-08:
Der Vollständigkeit halber noch die logs der warp von heute:
[event_log.txt](https://github.com/Tinkerforge/warp-charger/files/6100593/event_log.txt)
[debug_report.txt](https://github.com/Tinkerforge/warp-charger/files/6100596/debug_report.txt)

andig, 2021-03-08:
Ich meinte die Fehler bei abgestecktem Auto. Log? An welcher Stelle?

22kW, 2021-03-08:
Die charger out of sync warning bei ausgestecktem Auto gibt es alle 10 Sekunden, z.B.
[lp-1  ] WARN 2021/03/08 10:32:40 charger out of sync
[lp-1  ] WARN 2021/03/08 10:32:50 charger out of sync
[lp-1  ] WARN 2021/03/08 10:33:00 charger out of sync

andig, 2021-03-08:
Bitte vollständiges Log dazu

22kW, 2021-03-08:
neues log, jetzt bei
[lp-1  ] WARN 2021/03/08 11:41:26 charger out of sync
[lp-1  ] WARN 2021/03/08 11:41:36 charger out of sync
[lp-1  ] WARN 2021/03/08 11:41:46 charger out of sync

[log.txt](https://github.com/Tinkerforge/warp-charger/files/6100787/log.txt)

rtrbt, 2021-03-08:
Auffällig im debug_report ist, dass du ziemlich oft die MQTT-Verbindung verlierst. Ich habe das mal ad-hoc geplottet, die Zeit ist in Sekunden. Interessant ist die feste Grenze bei ungefähr 3 Stunden 20 Minuten. Auch interessant: die Verbindungen werden mit einem TCP Abort oder Reset geschlossen, also will die Gegenseite nicht mehr mit der Wallbox reden. Ich vermute aber, dass das nichts mit den EVCC-Meldungen zutun hat, dafür sind die Abbrüche zu selten.
[image]

rtrbt, 2021-03-08:
Teste mal bitte den aktuellen Stand des Pull Requests mit der angehangenen Firmware.
[warp_firmware_1_1_1_60463554_merged.bin.zip](https://github.com/Tinkerforge/warp-charger/files/6102016/warp_firmware_1_1_1_60463554_merged.bin.zip)

22kW, 2021-03-08:
OK:
(warp autostart steht auf AN (default)): 
Nach dem Start von evcc, solange contactor delay läuft, keine Warnung.
Danach wie gehabt:
[lp-1  ] WARN 2021/03/08 16:08:53 charger out of sync
Wenn ich dann im warp gui autostart auf off stelle verschwindet die Warning.

[2021-03-08-16-07-41-evcc-0.48-(67489e3).txt](https://github.com/Tinkerforge/warp-charger/files/6102282/2021-03-08-16-07-41-evcc-0.48-.67489e3.txt)

andig, 2021-03-08:
@rtrbt ich denke ich sehe das Problem:

    [lp-1  ] DEBUG 2021/03/08 16:08:43 charger status: A
    [lp-1  ] DEBUG 2021/03/08 16:08:43 next soc poll remaining time: 18m59s

-> evcc will abschalten

    [lp-1  ] DEBUG 2021/03/08 16:08:43 charger disable
    [warp  ] TRACE 2021/03/08 16:08:43 send /warp/evse/stop_charging: 'null'

-> sendet stop

    [lp-1  ] DEBUG 2021/03/08 16:08:43 charger disable
    [warp  ] TRACE 2021/03/08 16:08:44 recv /warp/evse/state: '{"iec61851_state":0,"vehicle_state":0,"contactor_state":1,"contactor_error":0,"charge_release":0,"allowed_charging_current":9000,"error_state":0,"lock_state":1,"time_since_state_change":1245414,"uptime":1245617}'

-> dennoch ist `charge_release` 0, also "enabled".

    [lp-1  ] DEBUG 2021/03/08 16:08:53 charge power: 3W
    [influx] TRACE 2021/03/08 16:08:53 write chargePower=3 (map[loadpoint:Garage])
    [influx] TRACE 2021/03/08 16:08:53 write chargeCurrent=0 (map[loadpoint:Garage])
    [lp-1  ] DEBUG 2021/03/08 16:08:53 charger status: A
    [influx] TRACE 2021/03/08 16:08:53 write chargedEnergy=0 (map[loadpoint:Garage])
    [lp-1  ] DEBUG 2021/03/08 16:08:53 next soc poll remaining time: 18m49s
    [lp-1  ] WARN 2021/03/08 16:08:53 charger out of sync

-> daher ab hier out of sync

    [warp  ] TRACE 2021/03/08 16:08:53 send /warp/evse/stop_charging: 'null'

Problem mit dem `charge_release`?

andig, 2021-03-08:
Beim zweiten drauf schauen glaube ich, dass das Problem das autostart ist. Für evcc muss das zwingend abgeschaltet sein. Ich würde das jetzt 1x zum Start setzen und eine Warnung ausgeben. Danach ist es Anwenderverantwortung, das es dabei bleibt.

andig, 2021-03-09:
Ist implementiert, bitte nochmal testen.

rtrbt, 2021-03-09:
Für mein Verständnis: "evcc will abschalten" heißt du willst die Ladefreigabe dauerhaft wegnehmen? Dann solltest du stop_charging schicken und auto_start_charging false gleich hinterher. Das spart soweit ich das gerade sehe dass du dich auf den Anwender verlassen musst.

Noch eine Frage: Wie funktioniert bei dir die Detektion ob die Box neugestartet ist? (Ist eher esoterisch aber es gibt ein paar Kunden, die die Box mit einem Rundsteuergerät abschalten wollen) Hintergrund ist, dass auto_start_charging nicht im Flash des EVSEs gespeichert wird. D.h. du müsstest das beim Start der Box neusetzen, nicht beim Start von EVCC.

andig, 2021-03-09:
> Für mein Verständnis: "evcc will abschalten" heißt du willst die Ladefreigabe dauerhaft wegnehmen?

Ja. Evcc entscheidet ob geladen wird, keine Automatik.

> Dann solltest du stop_charging schicken und auto_start_charging false gleich hinterher.

Genau: https://github.com/andig/evcc/pull/718/commits/166f71ccf751ad4a2cb3fd0a444975cfb1f325aa

> Wie funktioniert bei dir die Detektion ob die Box neugestartet ist? 

Gar nicht, wird über den gleichen Mechanismus mit angefangen.

rtrbt, 2021-03-09:
Ah ich hätte beim Diff den Kontext über der 142 aufklappen sollen, dachte das wäre in deiner Init-Funktion, aber das ist ja in enable.

22kW, 2021-03-09:
Neues log (Ich habe bei 2021/03/09 08:46:11 manuell im warp gui auto start ausgeschaltet)
[warp  ] TRACE 2021/03/09 08:46:11 recv /warp/evse/auto_start_charging: '{"auto_start_charging":false}'

evcc kann das wohl beim start nicht ausschalten, obwohl es das versucht
[warp  ] WARN 2021/03/09 08:43:25 auto_start_charging must be disabled
[warp  ] TRACE 2021/03/09 08:43:25 send /warp/evse/auto_start_charging: '{ "auto_start_charging": false }'
[warp  ] TRACE 2021/03/09 08:43:25 send /warp/evse/stop_charging: 'null'
[warp  ] TRACE 2021/03/09 08:43:25 recv /warp/evse/auto_start_charging: '{ "auto_start_charging": false }'
das hat autostart im warp GUI aber nicht verändert.

[2021:03:09-08-42-23evcc0.48(166f71c).txt](https://github.com/Tinkerforge/warp-charger/files/6106844/2021.03.09-08-42-23evcc0.48.166f71c.txt)

rtrbt, 2021-03-09:
Ah ja, @andig du triffst da das klassische MQTT-Problem: Das Topic auf dem man eine Konfiguration aktualisiert hat einen anderen Namen als das, auf dem die Konfiguration ausgegeben wird. Ansonsten würde die Wallbox sich selbst permanent Konfigurationsupdates geben, man kann ja bei MQTT nicht den Absender unterscheiden.

Du musst also evse/auto_start_charging**_update** aufrufen.

andig, 2021-03-09:
Oh yeah, ist behoben!

22kW, 2021-03-09:
jetzt geht es, autostart wurde von evcc ausgeschaltet

andig, 2021-03-09:
Klingt gut. Damit ready for merge oder fehlt nochwas?

22kW, 2021-03-09:
habe jetzt das Auto angestöpselt, und bekomme jetzt beim Laden charger out of sync

andig, 2021-03-09:
Logfile?

22kW, 2021-03-09:
[lp-1  ] WARN 2021/03/09 10:30:00 charger out of sync

[2021:03:09 10-17-58 evcc 0.48 (22478cd).txt](https://github.com/Tinkerforge/warp-charger/files/6107508/2021.03.09.10-17-58.evcc.0.48.22478cd.txt)

andig, 2021-03-09:
Nach Rücksprache mit @rtrbt nochmals umgebaut.

22kW, 2021-03-09:
es wird etwa dubios, jetzt schaltet evcc autostart auch selber wieder an:

[warp  ] TRACE 2021/03/09 11:45:02 send /warp/evse/auto_start_charging_update: '{ "auto_start_charging": false }'
[warp  ] TRACE 2021/03/09 11:45:02 send /warp/evse/auto_start_charging_update: '{ "auto_start_charging": false }'
[warp  ] TRACE 2021/03/09 11:45:02 send /warp/evse/stop_charging: 'null'
[lp-1  ] DEBUG 2021/03/09 11:45:02 max charge current: 16
[warp  ] TRACE 2021/03/09 11:45:02 send /warp/evse/current_limit: '{ "current": 16000 }'
[warp  ] TRACE 2021/03/09 11:45:02 send /warp/evse/current_limit: '{ "current": 16000 }'
[lp-1  ] DEBUG 2021/03/09 11:45:02 charger enable
[cache ] TRACE 2021/03/09 11:45:02 chargeCurrent: 0
[warp  ] TRACE 2021/03/09 11:45:02 send /warp/evse/auto_start_charging_update: '{ "auto_start_charging": true }'
[warp  ] TRACE 2021/03/09 11:45:02 send /warp/evse/auto_start_charging_update: '{ "auto_start_charging": true }'

und es gibt wieder [lp-1  ] WARN 2021/03/09 11:43:32 charger out of sync

[2021:03:09 11-42-20 evcc 0.48 (253b984).txt](https://github.com/Tinkerforge/warp-charger/files/6107955/2021.03.09.11-42-20.evcc.0.48.253b984.txt)

rtrbt, 2021-03-09:
> es wird etwa dubios, jetzt schaltet evcc autostart auch selber wieder an
Das ist soweit korrekt, so funktioniert jetzt die Ladefreigabe.

andig, 2021-03-09:
@rtrbt schau mal bitte:

    [warp  ] TRACE 2021/03/09 13:04:26 send warp/aussen/evse/auto_start_charging_update: '{ "auto_start_charging": false }'
    [warp  ] TRACE 2021/03/09 13:04:26 send warp/aussen/evse/stop_charging: 'null'
    [lp-1  ] TRACE 2021/03/09 13:04:26 charge currents: [1 2 3]A
    [warp  ] TRACE 2021/03/09 13:04:27 recv warp/aussen/evse/state: '{"iec61851_state":0,"vehicle_state":0,"contactor_state":1,"contactor_error":0,"charge_release":1,"allowed_charging_current":6000,"error_state":0,"lock_state":1,"time_since_state_change":80961350,"uptime":80961553}'
    [site  ] DEBUG 2021/03/09 13:04:27 ----
    [site  ] DEBUG 2021/03/09 13:04:27 pv power: -5800W
    [site  ] DEBUG 2021/03/09 13:04:27 grid power: 6400W
    [site  ] TRACE 2021/03/09 13:04:27 grid currents: [10 20 30]A
    [site  ] DEBUG 2021/03/09 13:04:27 site power: 6400W
    [lp-1  ] DEBUG 2021/03/09 13:04:27 charge power: 2800W
    [lp-1  ] DEBUG 2021/03/09 13:04:27 charger status: A
    [lp-1  ] WARN 2021/03/09 13:04:27 charger out of sync: expected disable got enable

Es sieht so aus, als würde der Schreibvorgang auf `auto_start_charging_update` nicht wieder zurück reflektiert. Kann es sein, dass da noch was in der FW fehlt?

rtrbt, 2021-03-09:
Siehe Mail, bei mir scheint es zu tun.

andig, 2021-03-09:
@22kW bitte nochmal probieren, ein weiterer Fix ist dazu gekommen

22kW, 2021-03-09:
läuft ohne warnings

andig, 2021-03-10:
Yeah, dann merge ich das mal. Hier machen wir zu? Danke an @rtrbt für die Unterstützung!

22kW, 2021-03-10:
Super vielen Dank an euch !

rtrbt, 2021-03-17:
Die offizielle Firmware 1.2.0 ist jetzt veröffentlicht, mit dabei ist die API-Erweiterung für EVCC.

closed 2021-03-17 by rtrbt