Tinkerforge/warp-charger #3: EVSE: Contactor error 6

22kW, 2021-03-12:
Ich habe im log einen  EVSE: Contactor error 6 und EVSE: Error state 4
Was bedeutet das ?

[event-log-2021-03-12T17-23-09-634Z.txt](https://github.com/Tinkerforge/warp-charger/files/6131844/event-log-2021-03-12T17-23-09-634Z.txt)
[debug-report-2021-03-12T17-23-11-108Z.json.txt](https://github.com/Tinkerforge/warp-charger/files/6131846/debug-report-2021-03-12T17-23-11-108Z.json.txt)

rtrbt, 2021-03-16:
Moin,

Kurze Bestandsaufnahme: Das interessante ist der Contactor error 6. Contactor errors sind die Fehlerzustände der Schützprüfung, die 6 bedeutet, dass der Ladecontroller das Schütz abgeschaltet hat, aber sowohl davor, als auch dahinter Strom anliegt. (Siehe hier: https://github.com/Tinkerforge/evse-bricklet/blob/ebb169d39cbd873a041757bcbdd1822ae2c153b1/software/src/contactor_check.c#L89 )
Der Error state 4 ist dann dass der Ladecontroller abschaltet, weil die Schützprüfung diesen Fehler erzeugt hat.

Die Fehler sind eine Sekunde später wieder weg, das kann aber in echt deutlich schneller sein, die Fehlerzustände frage ich nur einmal pro Sekunde vom Ladecontroller ab.

Mich wundern folgende Dinge:
1. Du sagst, dass das eine unerwartete Ladeunterbrechung war. Die kann ich mir aber nicht durch den Fehler erklären, weil ja zuerst der Ladecontroller denken muss, dass er das Schütz abgeschaltet hat, damit er danach den Zustand "Davor Strom, Dahinter Strom" als ungültig betrachtet. Warum sollte der Ladecontroller aber einfach so die Ladung abbrechen? D.h. wenn nicht aus irgendeinem anderen Grund jemand abgeschaltet hat, sollte das nicht passieren können.
2. Selbst wenn jemand anderes die Ladung gestoppt hat, ignoriert der Ladecontroller die Schützprüfung für eine kurze Zeit, damit das Schütz Zeit zum Schalten hat. Man sollte es also nicht hinbekommen, dass der Controller direkt nach dem Schalten kurz einen ungültigen Zustand sieht. In deinem Log ist aber eine Sekunde später (lies: das nächste Mal als ich den Ladecontroller nach seinem Zustand frage) das Problem wieder weg. Eventuell zieht das Schütz zu langsam an.

Folgende Fragen dazu:
- Ist das noch öfter passiert oder nur das eine Mal?
- Könntest du dir erklären, was die Abschaltung ausgelöst hat? Also eventuell dass jemand den Knopf gedrückt hat, das Auto nur auf 80% laden wollte oder EVCC beschlossen hat deine PV-Anlage erzeugt zu wenig Strom o.Ä.?
(Das die Abschaltung passiert ist nicht das Hauptproblem, aber eventuell gibt das einen weiteren Hinweis)

Ich hole mir bei uns mal noch eine zweite und eventuell dritte Meinung ein.

22kW, 2021-03-16:
- Das ist bisher ein einmaliger Vorgang
- soweit ich mich erinnere war das zu einem Zeitpunkt wo ich mit der PV Regelung von evcc rumexperimentiert habe, d.h. die Ladeunterbrechung kann durchaus von evcc initiiert worden sein.

rtrbt, 2021-03-16:
Die zweite Meinung ergab folgendes: Vielleicht hattest du wirklich einfach Timing-Pech und das Schütz hat länger zum Abschalten gebraucht, als die Firmware des Ladecontrollers erwartet. Beobachte das bitte weiter und gib Bescheid, falls das noch öfter auftritt.

22kW, 2021-03-18:
o.k. ihr glaubt also, dass kein glitch in der timing Logik oder hardware dahinter versteckt ist?

Ist denn die pwm auf cp vorher aus, oder schaltet das Schütz unter Last ?

borg42, 2021-03-19:
Wenn du "StopCharging" aufrufst oder den Knopf drückst setzen wir schon zuerst das PWM zurück. Das EVSE selbst weiß aber nicht ob das Schütz noch unter Last steht. Je nachdem wie lange das Auto braucht zu reagieren kann es passieren das unter Last geschaltet wird.

Wenn das Auto selbst entscheidet mit dem Laden aufzuhören (es erhöht den Widerstand wieder) schalten wir innerhalb von 100ms das Schütz, wie es vom Standard vorgeschrieben wird. Laut Standard sollte das dann auch nicht mehr unter Last sein.

22kW, 2021-03-23:
Fehler ist nie wieder aufgetreten

22kW, 2021-03-24:
Heute erneut aufgetreten:

[event-log-2021-03-24T14-53-22-707Z.txt](https://github.com/Tinkerforge/warp-charger/files/6198162/event-log-2021-03-24T14-53-22-707Z.txt)

[debug-report-2021-03-24T14-53-24-338Z.json.txt](https://github.com/Tinkerforge/warp-charger/files/6198163/debug-report-2021-03-24T14-53-24-338Z.json.txt)

batti, 2021-03-25:
Kannst du uns telefonisch erreichen?

22kW, 2021-03-25:
Unter welcher Nummer?

batti, 2021-03-25:
https://www.warp-charger.com/ -> unter Kontakt

22kW, 2021-03-25:
Meldet sich keiner.

22kW, 2021-03-26:
Problem bleibt auch nach dem Unterbrechen der Stromversorgung gestern bestehen:
[event-log-2021-03-26T11-42-12-392Z.txt](https://github.com/Tinkerforge/warp-charger/files/6211571/event-log-2021-03-26T11-42-12-392Z.txt)
[debug-report-2021-03-26T11-42-13-765Z.json.txt](https://github.com/Tinkerforge/warp-charger/files/6211572/debug-report-2021-03-26T11-42-13-765Z.json.txt)

22kW, 2021-04-26:
Hm, error 6 jetzt auch mit der neuen Box...

[event-log-2021-04-26T20-05-53-115Z.txt](https://github.com/Tinkerforge/warp-charger/files/6379926/event-log-2021-04-26T20-05-53-115Z.txt)
[debug-report-2021-04-26T20-05-54-588Z.json.txt](https://github.com/Tinkerforge/warp-charger/files/6379927/debug-report-2021-04-26T20-05-54-588Z.json.txt)

rtrbt, 2021-04-27:
Sehr seltsam. Wir haben deine alte Box hier angeschlossen und konnten das Problem bisher auch einmal erzeugen. Seit dem gab es aber einige Ladevorgänge, wo das nicht passiert ist. Das andere Problem mit dem Schütz das kurz anzieht, dann abfällt und wieder anzieht, haben wir aber bisher noch nicht gesehen. Ich habe auch versucht das mit unserem Tester zu erzeugen (~ 50 mal das Laden gestartet und wieder gestoppt) und es ist nichts passiert. Ich melde mich nochmal, wenn es mehr Erkenntnisse gibt.

rtrbt, 2021-05-05:
Kurzes Update: Heute Morgen konnten wir auch das andere Problem mit dem Schütz auch reproduzieren.

22kW, 2021-07-06:
Gib es hier neue Erkenntnisse ?

Ich habe weiterhin diverse Errors 4 und 6:

[event-log-2021-07-06T10-32-56-471Z.txt](https://github.com/Tinkerforge/warp-charger/files/6769335/event-log-2021-07-06T10-32-56-471Z.txt)

[debug-report-2021-07-06T10-32-59-132Z.json.txt](https://github.com/Tinkerforge/warp-charger/files/6769340/debug-report-2021-07-06T10-32-59-132Z.json.txt)

borg42, 2021-07-07:
Wir haben in der EVSE Firmware Version 2.0.9 Änderungen die Glitches in der Schütz-Fehler-Erkennung verhindern sollen. Das wird mit der nächsten WARP Firmware in Kürze veröffentlicht!

rtrbt, 2021-07-19:
Teste mal bitte [Firmware 1.2.4](https://www.warp-charger.com/#firmware). Damit sollten die Schützfehler-Glitches behoben sein.

22kW, 2021-07-28:
Sieht bisher gut aus, keinen Fehler mehr im Log mit Firmware 1.2.4

closed 2021-10-07 by 22kW