Tinkerforge/warp-charger #4: Irreguläres Verhalten vom Schütz bei Ladebeginn

22kW, 2021-03-18:
Heute (wieder) folgendes Verhalten:
Zoe eingesteckt, Schütz zieht für ca. 100ms an, fällt sofort wieder ab.
Nach ca. 3 Sekunden zieht es endgültig an und Auto lädt. 

evcc war NICHT aktiv, aber MQTT in der Box war noch enabled
Warp steht auf Autostart

Dies unstetige Verhalten vom Schütz nach dem Einstecken hatte ich auch vorher schon, habe es aber auf evcc geschoben. Diese Fehlerquelle scheidet hier jetzt aus.

rtrbt, 2021-03-18:
Moin,

Ich baue gerade ein Lade-Debug-Modul für die Firmware, eventuell finden wir damit raus, woran es hängt.

Als Nebeneinwurf: EVCC setzte bis vor 15 Minuten alle MQTT-Nachrichten auf retained (https://github.com/andig/evcc/commit/454688ac9f2974493f77c12e49d60746f621e84d), auch stop_charging und start_charging. Den Effekt von Nachrichten die retained werden, siehst du sogar nach einem Brokerneustart noch, weil die Broker die Nachrichten auf Platte speichern.
Ich glaube nicht dass das einen Zusammenhang hat, aber deaktiviere sicherheitshalber mal MQTT, starte dann die Box neu und sieh nach ob es dann immer noch passiert.

22kW, 2021-03-18:
>>deaktiviere sicherheitshalber mal MQTT, starte dann die Box neu 
habe ich gemacht im log steht:
34864 Client reconnected! Last message ID that it got is: 33597
Bezieht sich das client auf MQTT ? weil das hatte ich ja disabled

47              **** TINKERFORGE WARP Charger V1.2.0-6050aa83 ****
48                   310K RAM SYSTEM   283664 HEAP BYTES FREE
48          READY.
559         WARP Charger SPIFFS version 1.0.2-60213214
968         mDNS responder started
1286        Connecting to MinschMeier
1298        Had to configure softAP ip 2 times.
2169        Connected to MinschMeier
2198        Got IP address: 192.168.178.34
3298        Soft AP started.
3298            SSID: warp-SCt
3299            hostname: warp-SCt
4035            IP: 10.0.0.1
4046        Wifi connected. Stopping soft AP
34864       Client reconnected! Last message ID that it got is: 33597

rtrbt, 2021-03-18:
Nein, das ist dein Browser der sich wieder auf die Wallbox-Events verbunden hat.

rtrbt, 2021-03-18:
Nimm mal die angehangene Firmware. Auf der Ladecontroller-Unterseite kannst du jetzt ein Ladeprotokoll erstellen: Wenn du start drückst fängt die Aufzeichnung an, danach kannst du den Zoe anschließen, in der Hoffnung, dass das Problem wieder auftritt. Mit "stop + download" kannst du dann das Log abrufen. Das Log wird in deinem Browser aufgesammelt, also nicht den Tab währenddessen schließen.
Du solltest das nicht permanent laufen lassen, sonst wird das Log sehr länglich: Ich polle da alle 50 ms den Zustand des EVSEs.
[warp_firmware_1_2_0_605376c1_merged.bin.zip](https://github.com/Tinkerforge/warp-charger/files/6165493/warp_firmware_1_2_0_605376c1_merged.bin.zip)

22kW, 2021-03-19:
bekomme nach hochladen "wrong magic byte":
warp_firmware_1_2_0_605376c1_merged.bin.zip

rtrbt, 2021-03-19:
Du musst die Zip erst entpacken, ich habe nur gezippt weil Github es nicht erlaubt bin-Dateien hochzuladen.

22kW, 2021-03-19:
äh hups.
Habe es jetzt mit neuester Firmware und neuestem andig/evcc@454688a getestet, konnte das Problem nicht reproduzieren...

22kW, 2021-03-23:
Problem heute wieder aufgetreten (in Verbindung mit evcc), leider ohne Aufzeichnung. Mit Aufzeichnung konnte ich nicht reproduzieren...

22kW, 2021-03-23:
So, jetzt habe ich einen Mitschnitt:

[evse-debug-log-2021-03-23T15-02-45-120Z.txt](https://github.com/Tinkerforge/warp-charger/files/6190596/evse-debug-log-2021-03-23T15-02-45-120Z.txt)

22kW, 2021-03-23:
An power ist erkennbar, dass das Schütz anzieht, obwohl der contactor_state sich nicht ändert:

[image]

rtrbt, 2021-03-23:
Hilft diese Firmware? Ich habe die Theorie, dass das Schütz zu schalten länger EMI erzeugt als wir glauben und bei dir deshalb manchmal die Widerstandsmessung verfälscht wird, was zu dem Abschalten führt.
[warp_firmware_1_2_1_605a096a_merged.bin.zip](https://github.com/Tinkerforge/warp-charger/files/6190854/warp_firmware_1_2_1_605a096a_merged.bin.zip)

Bezüglich deines Graphen: contactor_state wird direkt nach dem Schalten kurzzeitig nicht aktualisiert, da wir den Zustand des Schützes auch über einen GPIO lesen. (Der durch den Schaltvorgang eventuell auch gestört wird). Prinzipiell wird der contactor_state nicht synchron mit dem Rest aktualisiert.

22kW, 2021-03-23:
Das Problem ist, dass es nur sporadisch Auftritt und ich leider auch nicht dauernd testen kann. Habt ihr einen Testaufbau mit Auto ? Vielleicht liegt es auch am Verhalten vom Renault Zoe. Ist allerdings bei anderen Wallboxen noch nicht aufgetreten.
Werde die firmware einspielen und beobachten...

rtrbt, 2021-03-24:
Wir haben hier zwei Boxen hängen an denen wir regelmäßig Zoes (ZE 40 und ZE 50 von 2019) laden, bisher ist niemandem sowas aufgefallen.

Ich könnte dir ein Python-Script schreiben, das du (z.B.) auf dem Rechner ausführen kannst auf dem du EVCC laufen hast, das automatisch Ladelogs erstellt, ohne dass du das im Browser aktivieren musst. Damit die Größe nicht explodiert, würde ich bis 10 Sekunden nachdem die Box auf State C gegangen ist loggen. Besteht da Interesse oder willst du das weiter von Hand machen?

closed 2021-04-19 by 22kW