Tinkerforge/warp-charger #5: Anzeige GPIO Low-Level-Zustand flackert

22kW, 2021-03-18:
Ist das unstete Verhalten der GPIO Anzeige beabsichtigt? 

[image]

https://user-images.githubusercontent.com/3041121/111594628-956d2100-87cb-11eb-8e97-8f2ecb2d534f.mov

borg42, 2021-03-18:
"Ausgang" ist die LED, da wird Low angezeigt wenn das PWM des atmen unter 50% ist high wenn es über 50% ist
"Motoreingangsschalter" ist auf dem EVSE ein Eingang für Wallboxen mit integrierter Buchse die einen Motor haben um das Kabel zu verriegeln. Dieser Eingang ist beim WARP Charger ungenutzt und floating und er kann entsprechend beliebige Werte anzeigen.

22kW, 2021-03-18:
aha, dann würde ich aber einen pull-up oder pull-down Widerstand für den Eingang empfehlen. Evtl. kann man das je nach Mikrocontroller auch intern aktivieren.

borg42, 2021-03-18:
Ah sorry, ich wurde gerade von meinem Kollegen korrigiert. Bei Wallboxen ohne Motorverriegelung (wie dem WARP Charger) wird eine Diode angebracht zwischen der LED und dem "Motoreingangsschalter". Anhand dieser Diode wird erkannt dass die Wallbox ein angeschlagenes Kabel hat. Dadurch sind "Ausgang" und "Motoreingangsschalter" so gekoppelt.

22kW, 2021-03-18:
ok, für den Laien sah das nur alles sehr floating aus, aber wenn es so seine Richtigkeit hat ist ja gut.

closed 2021-03-18 by 22kW