Tinkerforge/warp-charger #7: Zugangsdaten

elxsi, 2021-06-02:
Hallo zusammen,

ich habe heute den WARP Charger Pro installiert und wollte eben Zugangsdaten über die Web-Oberfläche konfigurieren (Firmware Version ist die 1.2.2-609a7746).

Enable Authentication ist angehackt
Username: test
Passwort: test

Das Form wird erfolgreich validiert (alles grün), ein Klick auf den Save Button bringt allerdings keine Rückmeldung. Die Zugangsdaten sind aber danach nicht gesetzt (getestet über ein neues private window im Firefox sowie über einen CURL Request über die API).

Ich habe auch versucht die Zugangsdaten per HTTP-API zu setzen:

curl -H "Content-Type: application/json" -X PUT -d "{\"enable_auth\": true, \"username\": \"test\",\"password\": \"test\"}" http://10.0.0.1/authentication/config
404

Hier bekomme ich aber nur ein 404 zurück. Ein Abruf klappt allerdings, zeigt aber auch, dass die Zugangsdaten nicht konfiguriert sind:

curl http://10.0.0.1/authentication/config
{"enable_auth":false,"username":"","password":null}

Was mir sonst noch aufgefallen ist:

Unter dem Menupunkt System ist bei dem ersten Eintrag (Logs) lediglich ein Icon aber keine Beschriftung. Wenn man den Eintrag öffnet gibt es zwei blaue Buttons ohne Beschriftung.

Unter System->Firmware update hat der blaue Button ebenfalls keine Beschriftung.

Danke!

rtrbt, 2021-06-07:
Moin,

Weil das besser passt, von hinten nach vorne:

>Unter dem Menupunkt System ist bei dem ersten Eintrag (Logs) lediglich ein Icon aber keine Beschriftung. Wenn man den Eintrag öffnet gibt es zwei blaue Buttons ohne Beschriftung.
>
>Unter System->Firmware update hat der blaue Button ebenfalls keine Beschriftung.

Die englische Übersetzung ist an der Stelle unvollständig bzw. passt nicht mehr zu den Platzhaltern die ersetzt werden. Ich baue mal einen Check, der die Übersetzungen beim Bauen der Firmware auf Vollständigkeit prüft. Schonmal danke für den Hinweis :)
Interessant ist aber, dass du überhaupt die englische Version des Webinterfaces bekommst. Hast du deinen Browser entsprechend konfiguriert oder ist das unbeabsichtigt?

>Ich habe auch versucht die Zugangsdaten per HTTP-API zu setzen:
>
>curl -H "Content-Type: application/json" -X PUT -d "{"enable_auth": true, "username": "test","password": "test"}" http://10.0.0.1/authentication/config
>404
>
>Hier bekomme ich aber nur ein 404 zurück.

Das sollte funktionieren, wenn du http://10.0.0.1/authentication/config_update verwendest. Ist hier dokumentiert:
https://www.warp-charger.com/api.html#states (unter Konfigurationen)
Dass du _update anhängen musst hat den Hintergrund, dass die Pfade für HTTP und MQTT gleich sind. Über MQTT würde die Box aber, wenn sie die Konfiguration auf dem selben Topic publisht, auf dem sie auch auf Konfigurationsupdates wartet, permanent mit sich selbst reden.

Ich muss mir mal ansehen, ob das für die HTTP-API separat trotzdem erlaubt sein kann, auf jeden Fall kann da aber eine bessere Fehlermeldung zurückkommen.

>Das Form wird erfolgreich validiert (alles grün), ein Klick auf den Save Button bringt allerdings keine Rückmeldung. Die Zugangsdaten sind aber danach nicht gesetzt (getestet über ein neues private window im Firefox sowie über einen CURL Request über die API).

Das habe ich kaputt refaktorisiert. Der Fix dafür kommt in der nächsten Version, voraussichtlich diese Woche.

elxsi, 2021-06-08:
Hey,

so vielen Dank erstmal für deine Rückmeldung.

> Moin,
>
> Weil das besser passt, von hinten nach vorne:
>
>
> Die englische Übersetzung ist an der Stelle unvollständig bzw. passt nicht mehr zu den Platzhaltern die ersetzt werden. Ich baue mal einen [...]

Gerne :)

> Interessant ist aber, dass du überhaupt die englische Version des Webinterfaces bekommst. Hast du deinen Browser entsprechend konfiguriert oder ist das unbeabsichtigt?

Das ist beabsichtigt, ich habe prinzipiell immer alles auf Englisch eingestellt, weil die deutschen Übersetzungen teilweise einfach furchtbar sind.

> 
> > Ich habe auch versucht die Zugangsdaten per HTTP-API zu setzen:
> > curl -H "Content-Type: application/json" -X PUT -d "{"enable_auth": true, "username": "test","password": "test"}" http://10.0.0.1/authentication/config
> > 404
> > Hier bekomme ich aber nur ein 404 zurück.
> 
> Das sollte funktionieren, wenn du http://10.0.0.1/authentication/config_update verwendest. Ist hier dokumentiert:
> https://www.warp-charger.com/api.html#states (unter Konfigurationen)
> Edit: Du musst danach noch einen Neustart auslösen, damit die geänderte Konfiguration verwendet wird.
> 
> Dass du _update anhängen musst hat den Hintergrund, dass die Pfade für HTTP und MQTT gleich sind. Über MQTT würde die Box aber, wenn sie die Konfiguration auf dem selben Topic publisht, auf dem sie auch auf Konfigurationsupdates wartet, permanent mit sich selbst reden.
> 
> Ich muss mir mal ansehen, ob das für die HTTP-API separat trotzdem erlaubt sein kann, auf jeden Fall kann da aber eine bessere Fehlermeldung zurückkommen.

Ah sorry, das hatte ich überlesen, danke für den Hinweis. Wenn ich _update anhänge klappt das soweit.  Ja eine bessere Fehlermeldung wäre schön, das 404 war etwas nichtssagend :)

> Das habe ich kaputt refaktorisiert. Der Fix dafür kommt in der nächsten Version, voraussichtlich diese Woche.

Super, danke!

rtrbt, 2021-06-11:
Firmware 1.2.3 ist veröffentlicht, Übersetzung und Authentication sollten jetzt repariert sein. Gib bitte nochmal Bescheid, wenn irgendetwas nicht funktioniert.

elxsi, 2021-06-11:
> Firmware 1.2.3 ist veröffentlicht, Übersetzung und Authentication sollten jetzt repariert sein. Gib bitte nochmal Bescheid, wenn irgendetwas nicht funktioniert.

So eben geflasht, scheint alles soweit zu funktionieren. Übersetzung sieht auch soweit gut aus jetzt. Danke sehr!

closed 2021-06-11 by elxsi