Tinkerforge/warp-charger #9: Missing end user documentation regarding firmware update

Alestrix, 2021-07-18:
The charger's manual states that on warp-charger.com new firmware can be found. This is incorrect.

Googling for the charger's firmware leads to this GitHub project, but the user is being left alone about where to find the firmware (can be found after some searching) and which of the two files - the large one or the "_merged" one - is to be uploaded to the wallbox. I haven't found this information until now and am hesitant to "just try it out" as I don't want to brick the box.

Request:
- Please add this info into the project's readme.
- Please add the link to the correct firmware into the respective "release" page (not only links to the source code)

bs-github, 2021-07-18:
You might have missed the link https://www.warp-charger.com/firmwares/warp_firmware_1_2_4_60f17413_merged.bin on the https://www.warp-charger.com/#firmware website.
I think it is quite obvious there.

Disclaimer: I'm as well just a customer.

KoljaWindeler, 2021-07-18:
Wasn't firmware one of the navigation item ? That seems to be changed ... Again ... Customer.

Alestrix, 2021-07-18:
Thanks for the pointers
I'm 99% sure the menu didn't look like this when I checked earlier today
[image]

I even went to warp-charger.com and did a text-search in the browser for "firmware" and nothing was found.

Having said that, I still think a pointer to the firmware would be a worthy addition to the readme. If I find the time I'll create a PR.

bs-github, 2021-07-18:
😂😂😂
Well, fancy dynamic web pages can be a UX nightmare and less predictable in terms of text searches. 

But google https://www.google.com/search?q=site%3Awww.warp-charger.com+firmware helps and 
the command line ```curl https://www.warp-charger.com/ | grep -i firmware``` works as well.

If you search for "firmware" on warp-charger.com within you browser right away there is nothing to be found 😟.
But if you scroll down first (or jump down via the menu), you will be pleased 😃.

[image]

KoljaWindeler, 2021-07-18:
Ha! The drop down item on the website depend on the subpage you're on ... See 
[image]
Fancy

rtrbt, 2021-07-19:
> Well, fancy dynamic web pages can be a UX nightmare and less predictable in terms of text searches.

Yes, the fancy animations unfortunately break searching text on the page. I've got a version of the main page on my disk without all this stuff, however I didn't have the time yet to clean the changes up and commit them.

> I'm 99% sure the menu didn't look like this when I checked earlier today

This sounds like a caching issue, the [changed menu structure](https://github.com/Tinkerforge/warp-charger/commit/3ed5530feb130ecc854af7f2116f71cf1793b394#diff-90e62cf319b4bd5379bc54c272743ecc96a660296f829274d58d1b9d42b5acf8) was commited on the 27th of April.

> Having said that, I still think a pointer to the firmware would be a worthy addition to the readme. If I find the time I'll create a PR.

No need for you to submit a PR, as I've just seen, the readme's content is still "TBD" only (since eight months!). I will add some content later today.

rtrbt, 2021-07-19:
Done.

closed 2021-07-19 by rtrbt