Tinkerforge/warp-charger #10: Einphasiges Laden

YuuLeeYian, 2021-08-07:
Hi,

ich habe heute einen warp-charger in betrieb genommen (Gute Nachricht; das Auto lädt 👍).

Nun wollte ich für das Überschussladen den Ladestrom über die API oder MQTT begrenzen. Dabei ist mir aufgefallen das der minimale Ladestrom auf **6000mA** über **3 Phasen** begrenzt ist (~4KW). Dies ist für die Solaranlage **viel zu viel**.

Für einen kleineren Ladestrom würde ich mir eine ein Phasen Ladung Konfiguration wünschen, um wenigstens auf ~1,4KW begrenzen zu können.

Es wäre auch gut wenn der Strom auf weniger als 6A zu begrenzen wäre.

Viele Grüße
Julian

borg42, 2021-08-07:
6A ist der kleinste Ladestrom der beim Laden über ein Typ2-Kabel möglich ist. Je nach Auto ist das Laden mit 6A allerdings sehr ineffizient. Bei einem Renault Zoe (z.B.) hat man erst ab 9A Ladestrom >90% Effizienz. Dadrunter sinkt die Effizienz rapide.

Der WARP Charger kann leider keine einzelnen Phasen abschalten, du könntest ihn aber direkt einphasig anschließen.

maugsburger, 2021-11-05:
Wäre es denkbar, ein weiteren zweipoligen Schütz (gerne mit Hilfskontakt, um das schon verbaute komplett abzuschalten) zu verbauen und anzuschließen?
Mit dem Dual Relay Bricklet müsste sich parallel eine CP Unterbrechung realisieren lassen, die einige Fahrzeuge zum Umschalten benötigen. Oder ist das auf dem EVSE schon möglich - leider sind die Schematics nicht eingecheckt als PDF zum schnell nachschlagen.

borg42, 2021-11-05:
Wir arbeiten aktuell an einem WARP Energy Manager, der zwei Phasen abschalten kann (und Energie messen für PV-Überschussladen). Dieser wird dann aber in der Unterverteilung angebracht und nicht in der Wallbox.

Das EVSE v2 im WARP2 hat bereits ein Relay zur CP-Trennung auf der Leiterplatte.

maugsburger, 2021-11-05:
Wird zwar leicht OT, aber heißt das im Umkehrschluss, dass die Aussage in der Broschüre 
https://raw.githubusercontent.com/Tinkerforge/warp-charger/master/documents/WARP2_Broschuere_low.pdf
> Low‐Level‐Daten wie CP/PE Widerstand, **phasenweise Strom, Spannung, Leistung, Leistungsfaktor, Phasenwinkel** u.v.m. auslesbar

sich nur auf die Pro-Variante bezieht, wenn der Energy-Manager das noch als eigene Funktion bekommen muss?

rtrbt, 2021-11-05:
Das phasenweise Auslesen von Strom, Spannung, usw. geht tatsächlich nur mit der Pro, weil nur in der Pro ein Zähler verbaut ist. Haben wir gerade für die neue Broschüre, die gleich in den Druck geht, gefixt. Danke für den Hinweis!

Alestrix, 2021-11-09:
> Das EVSE v2 im WARP2 hat bereits ein Relay zur CP-Trennung auf der Leiterplatte.

Habt Ihr Erfahrungen mit 1p3p Umschaltung und der aktuellen Zoe? Nach meinen letzten Infos misst sich die Zoe nur neu ein, wenn das Kabel abgezogen wird (PP Unterbrechung), aber das mag noch der Stand für ältere Modelle oder ältere Zoe-Software sein...

rtrbt, 2021-11-09:
Bei den Tests mit unseren Zoes hat die CP-Trennung tatsächlich den Unterschied gemacht. Ohne CP-Trennung gab es nur einen Fehlerzustand, mit CP-Trennung während der Phasenumschaltung hat es funktioniert.

Es gibt aber noch keine (veröffentlichte) API, mit der du die CP-Trennung des EVSEs auslösen kannst.

Alestrix, 2021-11-09:
In den einschlägigen Foren finden sich Berichte (bzw Berichte über Berichte) von Zoe-Besitzern, die sich durch 1p3p Umschaltung - angeblich trotz CP Unterbrechung - den OBC gegrillt haben. Die Behauptung war, dass die Zoe nur (erneut) prüft, ob 1p oder 3p anliegen (und bei 1p wohl N auf den L3 Eingang legt, um daraus eine Art 2-Phasen Ladung zu machen), wenn das Kabel voll abgezogen wird. Bleibt die N-L3-Brücke beim Zuschalten von L2+L3 bestehen, gibt's nen Kurzen.

Prüfen kann ich das alles leider nicht, aber ich denke, dass Ihr bei Euren Recherchen sicherlich auch über diese Themen gestolpert seid. Habt ihr dazu irgendwelche belastbareren technischen Fakten gefunden?

Und plant ihr, auch kundeneigene Schützschaltungen zu unterstützen oder nur die eigene? In meinem Fall existiert ein Shelly 3EM im Schaltschrank, der die Wallbox-Ströme, Spannungen, Wirkleistungen, cos phi und Gesamtleistung liefert und über dessen eingebautes Relais ich den L2/L3 Schütz ansteuern kann.

rtrbt, 2021-11-10:
Die Berichte kennen wir, da scheint aber extrem viel Hörensagen dabei zu sein. Das einzig halbwegs belastbare, was uns untergekommen ist, sind die Aussagen der openWB-Entwickler, beispielsweise hier:
https://openwb.de/forum/viewtopic.php?p=31505#p31505

Das bezieht sich alles soweit alles nur auf ältere Zoes, siehe auch hier:
https://openwb.de/forum/posting.php?mode=quote&f=4&p=36871

Vermutlich hat Renault das irgendwann gefixt. Wenn das Verhalten der alten Zoe wirklich war, wie man das so hört, wäre das nicht nur bei der Phasenumschaltung höchstgradig gefährlich: Da PP direkt im Stecker verkabelt ist, reicht es dann nicht, das Kabel am Wallbox-Ende abzuziehen, sondern man müsste z.B. beim Wechsel zwischen Wallboxen mit Dose, bei denen eine einphasig und eine dreiphasig lädt, den Stecker am Zoe abziehen.

Wir haben hier mit der Z.E. 50-Variante (Baujahr 2019/20) getestet, bei der die Phasenumschaltung mit CP-Unterbrechung geklappt hat. 

> Und plant ihr, auch kundeneigene Schützschaltungen zu unterstützen oder nur die eigene?

Das ist bisher noch unklar.

Alestrix, 2021-11-23:
Gibt es schon irgendwo öffentlich einsehbare Ideen, wie der **Energy Manager** umgesetzt werden soll?
- Wer steuert wen?
- Ist er als Konkurrenz/Alternative zu EVCC gedacht?
- Wie gut konfigurierbar sind die Schnittstellen zu den Nachbarkomponenten? Also z.B.:
   - Messpunkt am Hausanschluss
   - Abfrage SoC der Hausbatterie (falls vorgesehen)
   - 1p/3p Schützsteuerung
   -  ...
 - Sollen noch mehr Geräte statt der Wallbox damit gesteuert werden?

batti, 2021-11-26:
Wir in der nächsten Zeit dazu etwas offiziell veröffentlichen. Ich verlinke dies dann hier.

kaeferfreund, 2022-01-17:
Wenn ich das richtig interpretiere wird das Phasen abschalten mit der neuen ISO Norm überflüssig? https://github.com/Tinkerforge/warp-charger/issues/20

Davon abgesehen ist mir der Strom viel zu teuer um Ihn mit ineffizientem Laden zu verschwenden. Deswegen habe ich bei mir PV weiter aufgerüstet, dann kann man auch problemlos 3 phasig laden.

maugsburger, 2022-01-17:
> Davon abgesehen ist mir der Strom viel zu teuer um Ihn mit ineffizientem Laden zu verschwenden. Deswegen habe ich bei mir PV weiter aufgerüstet, dann kann man auch problemlos 3 phasig laden.

Schön dass du das Dach und die finanziellen Mittel dazu hast.

Trotzdem ist das Feature für Besitzur älterer BEVs und kleinerer Anlagen weiterhin sinnvoll. Laden mit 80% Effizienz ist noch immer günstiger als Einspeisen oder den Rest aus dem Netz zu ergänzen.

Alestrix, 2022-01-17:
> Laden mit 80% Effizienz ist noch immer günstiger als Einspeisen oder den Rest aus dem Netz zu ergänzen.

Zustimmung.
Zumal ich sehr skeptisch bin, wie diese Effizienzberechnungen alle zustandegekommen sind. Manchmal hab ich da den Eindruck, dass Wirkleistung und Wirkungsgrad gleichgesetzt werden.

kaeferfreund, 2022-01-19:
Ich möchte hier niemanden angreifen, der gerne einphasig laden möchte. Bitte nicht falsch verstehen.

Ich habe einfach folgendes für mich festgestellt:
  1. Wenn man ein Elektrofahrzeug als Hauptfahrzeug nutzt benötigt man ein Fahrzeug mit einer vernünftigen Reichweite. Dazu würde ich bei uns das Tesla Model 3 zählen. Das hat einen 60 kWh Akku (neues Standard Range + mit LFP Akku)
  2. Wenn ich diesen Akku mit 1,4 kWh lade lade ich einen leeren Akku 43 Stunden lang (100% Effizienz angenommen). Selbst wenn ich von dem doppelten ausgehe dauert es noch 16,5 Stunden. Die Sonne scheint im Schnitt pro Tag 4,6 Stunden. Ich lade also knapp 4 Tage. Irgendwann möchte ich das Auto aber auch benutzen... Bei mir ist das irgendwie unpraktikabel
  3. Egal welches reines EV ich bei uns lade, ich bekomme wenn ich einphasig lade "viel mehr Energie in den Akku". Stimmt natürlich nicht, das Laden ist einfach viel ineffizienter, denn die Reichweite die ich mit der Energie schaffe ist die gleiche wie bei der schnelleren 3-phasigen Ladung.
  4. Die Gründe sind ja relativ einfach: Der On-Board Lader hat einen Eigenverbrauch und im Winter muss die Akkuheizung laufen. Wenn die Heizung und das Ladgerät jetzt natürlich in Summe 43 Stunden läuft statt z.B. 6h müssen die Verluste deutlich größer sein.
  5. Wenn die Wärmepumpen nach und nach Verwendung finden hat man mit einer kleineren PV Anlage ohnehin kaum Überschuss.
  6. Wenn man es im Winter, Herbst, Frühling besonders effizient halten möchte kann man direkt nach der Ankunft laden oder so, dass bei Abfahrt fertig geladen ist, dass der Akku schon warm ist.

Bei begrenzter Dachfläche würde ich geringere Überschüsse immer in einen Akku laden und Abends verbrauchen. Damit hat man immerhin +90% Effizienz. Das lohnt sich bei den aktuellen Strompreisen auch wirtschaftlich.

Genug der Diskussion über den Nutzen,  wer einphasig laden will mit einem BEV soll es tun :)

Alestrix, 2022-01-19:
> Ich möchte hier niemanden angreifen, der gerne einphasig laden möchte. Bitte nicht falsch verstehen.
> 
> Ich habe einfach folgendes für mich festgestellt:
> 
> 1. Wenn man .... .....

Da sind halt schon ne Menge Wenns dabei - durchgängig Minimalladung, Wärmepumpe, Hausakku, BEV= HauptAuto.

Zum einen geht es beim 1p Laden ja darum, die **Möglichkeit** zu haben, auf 1.4kW herunter zu gehen. Das heißt ja noch lange nicht, dass das ein Dauerzustand ist. Meine Zoe hat nen 22kW OBC, somit kann ich in 1p von 6A bis 20A (wegen Schieflastgrenze) laden, was bei meiner 10kWp Anlage (Ost-West-Dach) perfekt ist. Zumal ist sie nur Zweitwagen und somit oft genug bei Sonnenschein an der Wallbox - zumindest im Sommer 🌞.

Auch ist ein Haus-Akku in vielen Fällen betriebswirtschaftlicher Wahnsinn. Ich habe mal mit meiner 2020er PV-Einspeisekurve (Bezug/Einspeisung des gesamten Jahres im 5-Minuten Raster) eine Simulation gemacht, was mir ein 10kWh Akku gebracht hätte - ganze 200€ Einsparung! Bei den aktuellen Akku-Preisen amortisiert sich das etwa in frühestens 30 Jahren - wenn Akku und Wechselrichter so lange durchhalten.

Es gibt daher durchaus sehr gute Gründe für das 1p Laden.