Tinkerforge/warp-charger #15: Make changes to evse/auto_start_charging persistent

sahni2021, 2021-12-02:
The current firmware re-enables the auto_start_charging after a restart of the controller i.e. when power was removed from the wallbox.
There is a need to make changes persistent in order to prevent unauthorised charging events. For example when an unattended WARP is easily accessible from the road but protected by allowing charging to be started only via NFC.

See this discussion for reference: [https://www.tinkerunity.org/topic/6779-ladestart-nur-durch-mqtt-oder-http](url)

photron, 2021-12-03:
Yes, this is on our TODO list already. Just not publicly tracked yet. This is a firmware change that is tracked here Tinkerforge/esp32-firmware#80 now.

closed 2021-12-03 by photron

rtrbt, 2022-04-07:
Fixed in 2.0.0.