Tinkerforge/warp-charger #17: Cryptographically sign firmware downloads + publish hashes

generalmanager, 2021-12-13:
To improve security of the upgrade path and simultaneously protect against corrupted downloads, please publish secure hashes of the firmware files. Specifically SHA256 and/or SHA512.

In addition it'd be great if you'd actually cryptographically sign your releases, e.g. with GPG so users can make sure they aren't uploading a compromised firmware to the charger, which often has priviliged network access as most users won't be able to jail it into a separate VLAN.

janknieling, 2023-10-20:
Das Feature mit dem Signieren der Firmware ist relativ einfach umzusetzen.
Der ESP32 bringt dafür bereits alle nötigen Features mit.
OTA sollte hier kein Problem sein der ESP32 bringt dafür bereits alles mit.
Man muss auf den ESP32 lediglich Secure Boot aktivieren und einen Publickey hinterlegen (und diesen natürlich vorher erzeigen). Dann lassen sich nur noch Firmwares auf den ESP laden, die mit diesem Key signiert wurden.
Die Doku gibt es hier von Espressif https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v2.html

bs-github, 2023-10-21:
Sorry @janknieling, der Hinweis auf die ESP32 secure boot Features geht hier total am Thema vorbei.

Es geht bei diesem issue um die Möglichkeit zu prüfen ob man eine bestimmte (unmodifizierte) Version der Firmware hat.
Genau das adressiert [Add firmware hashes. Fixes one part of #17](https://github.com/Tinkerforge/warp-charger/commit/768d4fdfb68e1339dc21396b33c64ef457fc67fa) bereits größtenteils.

Der Vorteil von open source ist ja genau die Offenheit.
Die Aktivierung von secure boot würde die Möglichkeit eine eigene selbst angepasste Firmware zu nutzen unnötig beschneiden.

borg42, 2025-07-13:
WARP Firmwares sind schon seit längerem signiert und eine Prüfsumme gibt es auch.

rtrbt, 2025-07-15:
Was noch fehlt ist, dass wir die Prüfsumme signieren (z.B. mit GPG) damit ein Nutzer (und nicht nur der ESP von einer Firmware zur nächsten) verifizieren kann, dass die Firmware nicht modifiziert wurde.