Tinkerforge/warp-charger #18: Ladevorgänge signieren

kaeferfreund, 2021-12-26:
Wir planen bei uns die Wallboxen für den öffentlichen Einsatz zu nutzen. Solange wir den Strom verschenken scheint es meiner laienhaften Auffassung nach keine Probleme zu geben, allerdings wird es schwieriger wenn man eine Vergütung für den Strom haben möchte.

Für eine eichrechtkonforme Abrechnung der Ladevorgänge ist es  zwingend notwendig einzelne Ladevorgänge zu signieren, so dass diese mit einer Transparenzsoftware z.B. https://github.com/SAFE-eV/transparenzsoftware geprüft werden können.

Gäbe es genug Interesse, dass ein PR zu dem Thema gemerged würde? Nur mal gefragt, bevor man sich hier die Arbeit macht?

kaeferfreund, 2021-12-26:
https://www.safe-ev.de/global/downloads/S.A.F.E.End-Nutzer-Handbuch_Transparenzsoftware1.1.0.pdf?m=1634105376&

https://oar.ptb.de/files/download/58d8ffad4c9184f55a2f94e3

borg42, 2021-12-27:
Natürlich sind wir da an einem PR interessiert!

Aus unserer Sicht sollte der WARP Charger dazu eine möglichst minimalistische Schnittstelle bereitstellen und das komplette Handling sollte auf dem Backend stattfinden.

Sprich: Eine zahlungspflichtige Landung wird z.B. über ein Handy auf dem Backend gestartet. Das Backend gibt die Ladung auf der Wallbox frei. Die Wallbox stellt dem Backend während und nach der Ladung nur die rechtlich notwendigen Daten zur Verfügung. Alles andere wird wieder vom Backend gehandhabt.

Dazu einmal die Frage in den Raum: Ist es stand aktuell wirklich rechtlich OK wenn die Wallbox selbst kein Display hat und kein Ausdruck für die Messung einer Ladung bereitstellt? Ich weiß das es da viel hin und her gab zu dem Thema und anfangs nicht ganz klar war was OK ist und was nicht.

Und noch eine Anmerkung: Ab Mitte nächsten Jahres wird wegen der neuen Ladesäulen-Verordnung noch zusätzlich ein EC- oder Kreditkartenlesegerät rechtlich notwendig. Spätestens da benötigt man dann neben dem WARP Charger zusätzliche Hardware.

borg42, 2021-12-27:
Oh, und ganz vergessen oben. Es müsste natürlich zusätzlich ein eichrechtskonformer Zähler eingesetzt und vom Backend ausgelesen werden. Der Zähler im WARP Charger Pro ist "nur" MID-konform und reicht daher nicht aus für öffentlich zahlungspflichtiges Laden.

Sprich: Im Endeffekt wäre der WARP Charger nur ein ferngesteuertes Schütz in einem Gehäuse und alles andere müsste sowieso extern geschehen?

kaeferfreund, 2022-01-16:
Wenn ich das richtig im Kopf habe reicht die Genauigkeit des MID Zählers aus und es braucht "lediglich" eine Manipulationssicherheit (das was die meisten mit Transparenzsoftware realisieren) sowie eine Zertifizierung damit der Zähler eichrechtskonform für das Laden von Batterieelektrischen Fahrzeugen wird?

janknieling, 2023-10-20:
Ich bin auch der Meinung, dass die MID Konformität des Zählers ausreichen müsste.
So weit ich das weiß sind ja die ganzen Stromzähler, die man vom Netzbetreiber zu Hause hat ja auch nicht mehr geeicht sondern „nur“ noch MID konform, was von der EU geschaffen wurde, um die Eichrechte EU weit zu vereinfachen.
Erst X Jahre nach dem Aufbringen der MID Konformität muss ja dann nachgeeicht werden.
Ob das bei öffentlichem Laden auch so ist weiß ich nicht ich würde aber mal sehr stark davon ausgehen…

borg42, 2025-07-13:
Ich schließe das Issue erstmal. Falls wir auf Dauer eine Eichrechtskonforme Wallbox anbieten wird diese auch signierte Ladevorgänge für das Transparenzregister übertragen können. Aktuell macht es aus unserer Sicht keinen Sinn.