Tinkerforge/warp-charger #19: Laden mit 1P zeigt 3 aktive Phasen an :(

kaeferfreund, 2022-01-09:
Hi zusammen,

ich habe leider seit einiger Zeit das Problem, dass beim Laden unserer Hybriden KFZs mit 1 Phase (das Kabel ist nur 1 phasig belegt, das Fahrzeug kann nur 1 phasig laden) in EVCC 4 kW angezeigt werden.

Wenn ich in den WARP Charger schaue wird hier die Wirkleistung mit 1,4 kW zwar richtig angezeigt, aber alle 3 Phasen als aktiv gezeigt. Hier geht also noch was schief...

Ab wann soll denn eine Phase als aktiv gelten in der Auswertung/Anzeige? 

[image]

borg42, 2022-01-10:
Wir hatten das gleiche Thema gerade im Forum: https://www.tinkerunity.org/topic/7269-einphasiges-laden-zeigt-stromverbrauch-f%C3%BCr-l2-und-l3-an/

Wir haben uns jetzt dazu entschieden den "active phase threshold" einfach zu erhöhen: https://github.com/Tinkerforge/evse-v2-bricklet/commit/7ee58086f3c9b76a374470d40e27a0633b174f78

Das Problem ist dann mit dem nächsten WARP2 Firmware-Release behoben!

closed 2022-01-10 by borg42

kaeferfreund, 2022-01-10:
Superb! Das ist schon mal eine gute Lösung.

Allerdings wäre es ja EVCC seitig sinnvoll immer alle Ströme auszuwerten oder? Dann hat man nicht nur eine genauere Information sondern auch das Problem nicht mehr, selbst wenn der threshold erneut überschritten wird.

borg42, 2022-01-10:
Ja, da müsstest du dann bei EVCC schauen warum die das so machen: https://github.com/evcc-io/evcc/

Ich kenne da die Implementierungsdetails nicht.

rtrbt, 2022-01-10:
Ping @andig

kaeferfreund, 2022-01-10:
Die tatsächlichen Ströme der Phasen sind ja via MQTT exposed oder übersehe ich da etwas?

rtrbt, 2022-01-10:
Ja, die sind Teil von meter/detailed_values: https://www.warp-charger.com/api.html#meter_detailed_values

andig, 2022-01-10:
@kaeferfreund evcc macht die Phasenerkennung über die Ströme anstatt das Warp API dafür nu nutzen. Im Bild oben müsste da 1p raus kommen (wir haben 1A als Erkennungsschwelle definiert). Wenn das bei Dir nicht klappt mach bitte mal ein Logfile mit `--log trace` oder

    log:
      warp: trace

so dass ich sehen kann, was da über MQTT kommt.

> ich habe leider seit einiger Zeit das Problem, dass beim Laden unserer Hybriden KFZs mit 1 Phase (das Kabel ist nur 1 phasig belegt, das Fahrzeug kann nur 1 phasig laden) in EVCC (Version 0.79) 4 kW angezeigt werden.

Deine Warp hat einen Zähler? Dann sollten wir egtl. gar nicht mit den Strömen rechnen (3p x 6A) sondern einfach den Wert vom Zähler nehmen. Auch warum das nicht klappt bekommen wir mit dem Logfile raus. 

Details gerne hier oder in neuem Ticket bei evcc.

kaeferfreund, 2022-01-10:
Klasse, vielen Dank!

rtrbt, 2022-01-10:
> Das Problem ist dann mit dem nächsten WARP2 Firmware-Release behoben!

WARP2 1.1.2 ist veröffentlicht, damit sollten die aktiven Phasen jetzt passen.

kaeferfreund, 2022-01-10:
Mega! Vielen Dank!