Tinkerforge/warp-charger #20: Kompatibilität mit der ISO 15118

kaeferfreund, 2022-01-17:
Die neue ISO 15118-2 scheint auch für Heimanwender interessant zu sein, unter anderem ist es (anscheinend) möglich, dass das Fahrzeug den SoC oder zumindest die benötigte Energiemenge mit der Wallbox austauscht.

Da mir dir Norm derzeit nicht vorliegt wäre es auf jeden Fall interessant ob man das rein technisch unterstützen könnte?

borg42, 2022-01-21:
ISO 15118 wird mit dem Auto per CAN over Ethernet over PowerLAN über den ChargePilot-Pin gesprochen.
Die WARP Charger haben keine Hardware verbaut die das könnte (ist auch leider sehr teuer).

kaeferfreund, 2022-01-21:
Oh krass, dann können die Autos das derzeit auch nicht?

borg42, 2022-01-21:
Die meisten Autos können das. Ohne ISO 15118 kann man soweit ich weiß nicht per DC laden.

Die allermeisten AC Wallboxen können aber kein ISO 15118.

Man bräuchte dazu so ein Modul hier: https://www.codico.com/de/white-beet-ei-evse-embedded-iso15118-module
Dazu dann noch ein bisschen Hardware um das Modul auch anzusteuern etc.

kaeferfreund, 2022-01-24:
Ah, dann sind die hybriden sehr wahrscheinlich nicht ISO 15118 kompatibel. Deswegen auch der CAN Bus der dann direkt mit 

Das Board schaut interessant aus. Anbindung via SPI ist schick. Wenn ich das richtig verstanden habe kann das Modul auch SAE J1772 und wäre dann ein Ersatz für das ESVE Modul? Alternativ Entwickelt man ein eigenes Modul auf Basis des ESVE, aber das kostet natürlich einiges an Zeit.

Auf jeden Fall spannend wo die Entwicklung hin geht, auch mit VW und der DC Wallbox...

janknieling, 2023-10-29:
@borg42 könnte man nicht auch so ein Modul, wie das hier https://www.codico.com/de/red-beet-e-1-1-evse-module in das  EVSE 2 Bricklet integrieren?
Das ist deutlich billiger und sollte meinem Verständnis nach auch ISO 15118-3 support liefern und das für ~30€

borg42, 2023-10-29:
Da fehlt die Software. Das andere Modul ist unter anderem so teuer weil das inkl. einer Lizenz für einen embedded ISO15118-Stack kommt.
Es gibt aktuell noch keinen Open Source embedded ISO15118-Stack und das selber zu implementieren liegt außerhalb unserer Möglichkeiten.

janknieling, 2023-10-29:
Ah interessant. Ich dachte da gibt es vllt mittlerweile was.
Ich bin auf der Suche nach einer Implementierung der ISO15118 über das Projekt hier gestoßen https://github.com/SwitchEV/iso15118 . Die bieten eine python Implementierung der ISO15118 die könnte man ja rein theoretisch auch auf dem ESP32 zum laufen bringen mit MicroPython

BrainGameFactory, 2024-11-04:
Es gibt ein Open Source Projekt was genau das macht. https://github.com/uhi22/foccci

borg42, 2024-11-04:
> Es gibt ein Open Source Projekt was genau das macht. https://github.com/uhi22/foccci

Das Projekt kennen wir gut, die implementieren aber die andere Seite. D.h. die bauen einen Controller mit dem man an DC-Ladesäulen laden kann. Gedacht für einen Umbau von Verbrenner auf Elektroauto.

christophwen, 2025-03-07:
@borg42 wäre dieses Open-Source-Projekt https://github.com/EcoG-io/iso15118 etwas?

Ich hab erst überlegt, mich mal in die Norm einzulesen, um hier einen besseren Kommentar zu hinterlaseen, als nur das Projekt zu verlinken. Ich durfte dann aber kennenlernen, wie teuer es ist, die einzelnen ISO Teile zu bekommen, um sie zu lesen. Ihr kennt euch da bestimmt schon viel besser aus. Wenn das was Passendes ist, dann könnte ich mir vorstellen hier mitzuarbeiten/zu unterstützen.

borg42, 2025-03-07:
Die machen schon das richtige, den Python-Code können wir aber natürlich leider nicht auf dem ESP32 laufen lassen.

Ich hab aber auch eine gute Nachricht: Wir sind schon um einiges weiter mit der ISO15118-Integration, wird auch in Kürze den ersten Code auf github geben, wo dann auch gerne mitgehofen werden kann :).

christophwen, 2025-03-07:
Ahh sehr cool, basiert die Implementierung auf den ISO Dokumenten, oder auf der Basis von einem anderen Projekt?

borg42, 2025-03-07:
Man schaut natürlich auch in andere Projekte rein um sich zu inspirieren, aber wir implementieren schon auf Basis der Norm.

peduepp, 2025-03-24:
Die Firma INSYS icom bietet ISO 15118 Kommunikationsmodule für die Hutschiene an: https://www.insys-icom.com/en/products/e-mobility-solutions/. Das Modul wird in die bestehende CP-Leitung geschaltet und verfügt über ein Ethernet-Interface. Eventuell eine interessante Option für die Nachrüstung?

borg42, 2025-03-24:
Danke für den Hinweis. Ich hab aber den ersten Prototypen-EVSE/ESP-Stapel mit ISO15118-Support schon bei mir auf dem Tisch liegen :).

borg42, 2025-03-26:
Der erste Code ist jetzt als feature-iso15118 branch im esp32-firmware git gepusht: https://github.com/Tinkerforge/esp32-firmware/tree/feature-iso15118/software/src/modules/iso15118

Der QCA700X Treiber, SLAC, SDP und DIN SPEC 70121 funktioniert damit rudimentär/prototypisch.

borg42, 2025-03-30:
> Kannst du schon mehr Hintergründe zur verwendeten Hardware teilen?

Wir haben im Moment Prototypen mit RED Beet und PLC Stamp Micro integriert auf dem ESP32 Brick. Was es am Ende wird steht noch nicht fest, hängt auch von den EMV-Tests ab die wir noch machen müssen.

Die Verbindung zwischen ESP und EVSE ist jetzt nicht mehr über ein Bricklet-Kabel, sonder über Board-To-Board-Stecker, da wir CP und eine Zero-cross detection benötigen für Powerline Communication,  was wir nicht über zusätzliche Kabel führen wollten. Das EVSE hat auch eine neue Stromversorgung bekommen, da wir ein bisschen mehr Strom benötigen für das PLC-Modem.

Ist also quasi einmal alles neu beim EVSE/ESP-Stapel, aber vom Formfaktor und von der Größe her noch genauso wie vorher.