Tinkerforge/warp-charger #21: Ladestrombegrenzung per MQTT setzen

rediculum, 2022-04-24:
Hallo zusammen

Ich konnte in der Doku nichts finden und ein Versuch einen Publish auf evse/max_charging_current mit dem entsprechenden JSON payload hat den Wert nicht verändert. Ist dies nicht implementiert oder mache ich etwas falsch? Ich fahre noch mit der 1.2.2 Firmware.

Danke und Gruss aus der Schweiz

rtrbt, 2022-04-25:
Moin,

Die API-Docs sind schon auf 2.0.0 umgestellt, deshalb kann ich das nicht verlinken, aber du musst auf evse/max_charging_current_update publishen. Prinzipiell alles, was du unter foo/bar lesen kannst musst du (falls die API das hergibt) unter foo/bar_update schreiben. Das ist so weil die Wallbox nicht unterscheiden kann, von wem auf ein Topic geschrieben wurde und sie sonst ihre eigenen Nachrichten wieder als Befehle interpretieren würde.

rediculum, 2022-04-25:
Moinsen :)

Mit max_charging_current_update funktioniert es leider auch nicht.

```
$ mosquitto_pub -t warp2/evse/max_charging_current_update -m '{"max_current_configured":10000,"max_current_incoming_cable":16000,"max_current_outgoing_cable":20000,"max_current_managed":32000}' -d
Client (null) sending CONNECT
Client (null) received CONNACK (0)
Client (null) sending PUBLISH (d0, q0, r0, m1, 'warp2/evse/max_charging_current_update', ... (130 bytes))
Client (null) sending DISCONNECT
$ mosquitto_sub -v -t 'warp2/evse/max_charging_current'
warp2/evse/max_charging_current {"max_current_configured":16000,"max_current_incoming_cable":16000,"max_current_outgoing_cable":20000,"max_current_managed":32000}
```

rtrbt, 2022-04-25:
Hm man sollte nicht Issues beantworten vor dem ersten Kaffee, sorry.
evse/max_charging_current beinhaltet alle Ladeströme, auch die, die man nicht setzen kann. Wirklich schreiben kannst du nur max_current_configured, dafür gibt es evse/current_limit (ohne _update, weil man da nur schreiben kann).

Die API ist an der Stelle leider etwas konfus, das wird besser wenn du auf die 2.0.0 updatest ;)

Falls du die alte Dokumentation noch benutzen willst kannst du dir das Git klonen und mit `git checkout 698c10b2` auf den Stand der alten Dokumentation wechseln.
Dann am besten im Ordner warp-charger.com einmal `python3 -m http.server` ausführen, dann kannst du unter http://localhost:8000/api.html auf die Doku zugreifen

rediculum, 2022-04-25:
Ich habe jetzt auf die FW 2.0.0 updated (nachdem ich vollgeladen hab, nur um sicher zu gehen 🤣 ) und so hab ichs jetzt rausgefunden:

Mittels folgendem Publish kann man den Ladestrom per MQTT updaten, welcher im UI unter "Configured charging current" gesetzt werden kann:

```
$ mosquitto_sub -v -t 'warp2/evse/global_current'
warp2/evse/global_current {"current":16000}
$ mosquitto_pub -t warp2/evse/global_current_update -m '{"current":10000}' -d
Client (null) sending CONNECT
Client (null) received CONNACK (0)
Client (null) sending PUBLISH (d0, q0, r0, m1, 'warp2/evse/global_current_update', ... (17 bytes))
Client (null) sending DISCONNECT
$ mosquitto_sub -v -t 'warp2/evse/global_current'
warp2/evse/global_current {"current":10000}
```

Der Issue kann geschlossen werden. Danke für die Hilfe und Grüsse

closed 2022-04-26 by rtrbt