Tinkerforge/warp-charger #23: UX: Tausender Trennzeichen nicht einheitlich

rediculum, 2022-05-05:
Hallo

Das 1000er Trennzeichen ist nicht überall einheitlich. Einmal ist es ein Punkt, dann ist es ein Komma oder gar nicht getrennt.

[image]

Ich persönlich bevorzuge Hochkommas als 1'000 Trennzeichen. Aber so wie es im Moment ist, sieht es für mich nicht schön aus.

Grüsse
Roland

rtrbt, 2022-05-06:
Moin Roland,

Mit welchen Spracheinstellungen und welchem Browser hast du den Screenshot gemacht? Ich bekomme gerade nicht erzeugt, dass Configured charging current mit . als Dezimaltrennzeichen ist und Allowed Charging current mit ,
Bezüglich der Hochkommata: Je nach Browser müsste das wenn du Deutsch (Schweiz) in den Spracheinstellungen wählst eigentlich funktionieren. Habe ich aber nicht getestet.

rediculum, 2022-05-06:
Der obige Screenshot is mit Chrome unter Windows. In den Spracheinstellungen des Browsers steht English US als erstes, deshalb die Website auf Englisch. Auch im Chrome unter Mac ist "Allowed Charging Current" mit Komma, jedoch mit Safari (auf Deutsch) ist ein Punkt. Jetzt kommts dicker. Unter Android Chrome (Egal ob DE_de/ch oder EN_us) hat auch das "Configured charging current" ein Komma, statt ein Punkt.
Es scheint also definitiv Browser- & OSabhängig zu sein und liegt nicht in der Programmierung.

Mir persönlich würde noch gefallen, wenn der 1000er Wert der Y Achse bei der Charge History (Im Status und im Energy Meter Menü) noch mit einem Zeichen getrennt wäre. Ob jetzt Komma, Punkt, Hochkomma ist mir Wurscht ;)

Gruss
Roland

closed 2025-07-13 by borg42