Tinkerforge/warp-charger #26: Ladecontroller Tastereinstellung nicht persistent

binarybucks, 2022-11-11:
Ich verwende den Taster der WARP2 Wallbox zum Schließen des Garagentores über MQTT. 
Hierzu ist die Tastereinstellung auf Keine Aktion gesetzt. 
Leider verliert die Wallbox sporadisch diese Einstellung und die Tastereinstellung wird auf den Default Ladestart bzw. Ladestop zurückgesetzt, so dass der Ladevorgang manchmal unerwartet gestoppt wird. 

Aktuell installierte Firmware: 2.0.8-63218f23

borg42, 2022-11-11:
Mh, kann es sein dass die Wallbox vom Strom getrennt wird sonst irgendwie neustartet? Die Tastereinstellung ist aktuell nicht persistent, d.h. bei einem Neustart wird sie wieder zurück gesetzt auf die Defaulteinstellung.

Vielleicht sollte die allerdings persistent sein.

binarybucks, 2022-11-11:
Eigentlich nicht. 
Laufzeit aktuell 11 Tage, 17:26:35. 
Danach wurde der Taster einmal neu gesetzt. Heute hatte er wieder seine Einstellung vergessen. 

Generell wäre es gut, wenn die Einstellungen auch nach einem Neustart persistent sind. Die anderen Einstellungen sind es ja auch, oder?

borg42, 2022-11-11:
Ja ich geb dir recht, die Einstellung sollte auch persistent sein, baue ich für die nächste Firmware-Version ein. Ich weiß gar nicht warum sie nicht persistent ist.

Allerdings ist es trotzdem komisch dass dein WARP Charger die Einstellung vergisst ohne einen Neustart :thinking:.

rtrbt, 2022-11-11:
> Laufzeit aktuell 11 Tage, 17:26:35.

Sicherheitshalber: Ist das die Laufzeit des Ladecontrollers (aus der entsprechenden Unterseite)?

binarybucks, 2022-11-11:
> Sicherheitshalber: Ist das die Laufzeit des Ladecontrollers (aus der entsprechenden Unterseite)?

Genau

rtrbt, 2022-11-25:
WARP2 Firmware 2.0.9, die ich gerade veröffentlicht habe, speichert die Einstellungen jetzt persistent. Falls die Einstellungen sich bei dir (vorallem ohne bewussten Neustart!) nochmal zurücksetzen, mach das Issue bitte wieder auf.

closed 2022-11-25 by rtrbt