Tinkerforge/warp-charger #28: Error: User not found, after deleting all and creating new ones

mkrasselt1, 2023-01-25:
Problem description:
after installation and upgrade to firmware 2.0.11-639c76c2 with configuration version 2.0.0 i wanted to enable authentication, but the default users were not of use for me. so i deleted all the rfidcards and all the users and saved. it asked me, if i wanted to restart or not, as those changes were not applied otherwise. Then i tried to add new users like admin with username and admin. so i clicked add user, enterd username: "warpadmin", name: "Administrator (lädt nie!)", limit: "16,000" and the password and clicked add. 
The system told me: "Error: No such user", but added the user anyway, but without my charge limit. And i could not save the data - but repeat it and add a second user with the same data - but still unable to save. i needed to save and restart befor i could add a new admin.

but this time, it wont save my password and my entered limit -> the limit needed to be enterd with the plus minus buttons (why, when you can enter directly) and the password needed to be disabled and the user saved before i could add a password and enable authentication. so thats 3 restarts in total - is it meant to bee like this? 

Stats:
- Warp2 Pro Charger 
- LAN Connection with MQTT active
- load management active

expected behavior:
- remove and add new users without restart
- able to insert ampere limit via numpad
- able to add a/multiple user with password set from the start

rtrbt, 2023-01-25:
Hi,

Can you please test this again with the current beta firmware? https://www.tinkerunity.org/topic/9215-warp2-beta-firmware-mit-modbus-tcp-wireguard-rtc-und-ocpp-unterst%C3%BCtzung/

There were some issues with how the API used to add/modify/delete users handles multiple calls in succession, and with the input handling of floats, however those should be fixed in the beta firmware. So all of

> - remove and add new users without restart
> - able to insert ampere limit via numpad
> - able to add a/multiple user with password set from the start

should work with the beta, and thus with the next "normal" firmware release.

> so thats 3 restarts in total - is it meant to bee like this?

The web interface always reminds you to restart after saving, however this is only necessary if you want to apply the changes.

mkrasselt1, 2023-02-10:
So either i have to upgrade all my chargers or disable load sheding for the time beeing?
Also i have a new set of errors with the latest stable firmware 2.0.11 - i cant edit the nfc tokes any more,
It says now "Aktuali­sieren der NFC-Ein­stell­ungen fehl­ge­schlagen Error: Timeout beim Herunter­laden" on 3 of 4 devices,
followed by a lost connection and what seams to be a restart :/ no obvious hints in the logfile, the js console or the networking log of my browser.

i will try that beta - but its unpleasant that the load management does not also include a combined nfc & user log/list - or could you add an options that those infos could be fetched via api from the master wallbox :) ?

rtrbt, 2023-03-23:
Fixed with 2.1.0.

closed 2023-03-23 by rtrbt