Tinkerforge/warp-charger #30: Version ohne integriertes Kabel mit Buchse/Steckdose

janknieling, 2023-03-05:
Ich fände es sehr sinnvoll, wenn es zusätzlich zu den beiden Versionen mit 5m und 7,5m Ladekabel mit CEE Stecker eine Version geben würde, die lediglich eine CEE Buchse anbietet anstatt dem Kabel.
Ich persönlich würde das bevorzugen, da eine Version mit Kabel durchaus Leute dazu animieren könnte das Kabel zu beschädigen und ich da auch Vorteile in Sachen Haltbarkeit sehe, da bei einer Version ohne fest angeschlossenes Kabel das Kabel nicht 24/7 der Witterung ausgesetzt wäre. 
Des Weiteren wäre so eine Box meiner Meinung nach optisch ansprechender und könnte doch bestimmt dann auch etwas kostengünstiger hergestellt werden, da man hier das Kabel spart.
Bei so gut, wie jedem E Auto oder Hybrid bekommt man ja ohnehin so ein Kabel dazu und hat es im Kofferraum liegen.

janknieling, 2023-03-06:
Meine Idee zur Umsetzung wäre, damit man das Gehäuse nicht komplett umdesignen muss, dass man wo aktuell das Typ 2 Kabel ausgeführt wird ein Kabel mit einer Typ 2 Buchse dran ausführt und rechts an der Box mit einer Halterung befestigt oder, dass es oben an der Oberseite der Box angebracht ist und da freischwebend auf dem Plastikgehäuse sitzt.

batti, 2023-03-09:
Die Idee ist sehr gut! Wir haben aber noch keine Lösung am Markt dafür gefunden. Es gibt abgesetzte Dosen die man direkt anschließen könnte. Diese Dosen verriegeln allerdings aber nicht, sprich der Verriegelungsmotor fehlt. Auch wenn andere Wallboxen ebenfalls nicht verriegeln, halten wir dies für notwendig. Einfach ein wasserdichtes Gehäuse zu nehmen und da eine Typ2 Dose mit Verriegelungsmotor einzusetzen gefiel uns bisher aus optischen Gründen nicht.

janknieling, 2023-03-09:
@batti das mit dem abgesetzten Gehäuse glaube ich gerne das ist optisch schwierig zu gestalten das müsste ja dann, damit es optisch passt auch wieder einen Edelstahldeckel haben. Die Frage ist, wie hoch hier der Aufwand wäre. Den Aufwand so ein Kunststoffgehäuse zu designen würde ich persönlich eher moderat einschätzen. Dann könnte man das ja sogar ggf. als Upgradekit anbieten, wo man das Gehäuse entweder kaufen oder selbst ausdrucken kann (mit dem richtigen Fillament sollte das ja wetterfest sein). Also eine Art Bausatz.
Ich hatte mich auch schon mal nach Typ 2 Dosen mit Verriegelungsmotor umgesehen um vllt selbst was zu bauen und einen PR zu machen.
Was mir noch einfallen würde ist, dass man die Steckdose in den Edelstahldeckel einsetzt. Dafür müsste man dann aber vermutlich das Kunststoffgehäuse entweder höher oder flächenmäßig größer machen, da da aktuell nicht genügend Platz drinne ist oder sehe ich das falsch?

batti, 2023-03-23:
Das siehst du richtig. Da ist ansonsten nicht genug Platz. Wir schauen mal was wir da zukünftig machen.

janknieling, 2023-10-21:
@batti hast du evtl einen Link oder ein Bild für mich, was für einen abgesetzten Stecker ihr da ins Auge gefasst hattet?
Ich bin gerade dabei mich mal am Markt umzusehen und evtl da was zu entwerfen.
Habe da nichts wirklich im Bereich der abgesetzten Typ 2 Buchsen finden können, was da passend aussieht.
Ich wollte mich mal dran setzen und was in der Richtung designed, wenn ich eine entsprechende Buchse und einen Verriegelungsmotor gefunden habe.
Dann könnte man sich ja so eine Box auch super als Mobile Wallbox für unterwegs hinlegen (mit CEE Stecker für den Anschluss ans Stromnetz)

batti, 2023-10-23:
Ich habe bisher noch keine abgesetzte Typ2 Dose mit Verriegelungsmotor gefunden. Durch den Motor wird die "Box" vergleichsweise groß. Wenn du diese Box als Mobile Wallbox nutzen möchtest, dann benötigst du die gesamte Technik mit Schütz etc.

janknieling, 2023-10-23:
Ich betrachte das beides aktuell eher als getrennte Projekte. zu der mobilen Box: Ich vermisse aktuell einfach am Open Source Markt etwas, dass in die Richtung hier geht: https://shop.go-e.com/go-e-charger-gemini-flex-22-kw
Vor allem stört mich an diesen ganzen Teilen, dass die alle in irgendeiner Form eine App haben, die dann garantiert nach Hause telefoniert…

Zum Thema abgesetzte Typ 2 Dose: Ich meinte da tatsächlich, ob du vllt was ohne Motor für mich hattest, damit ich sehen kann, was du dir da Designtechnisch vorgestellt hast. Vllt gibt es da ja eine Möglichkeit sowas sich mit Motor auszurüsten. Will mal ein bisschen basteln vllt bekomme ich da was hin habe da schon die eine oder andere Idee

tugsi, 2024-05-08:
Bezüglich verriegelter Typ2-Dosen, da hat Phoenix Contact einiges im Angebot:
[https://www.phoenixcontact.com/de-de/produkte/ladetechnik-fuer-die-elektromobilitaet/infrastruktur-ladedosen](https://www.phoenixcontact.com/de-de/produkte/ladetechnik-fuer-die-elektromobilitaet/infrastruktur-ladedosen)
Auch schon mit integrierter LED-Anzeige etc. 
Dies würde aber natürlich nicht mehr in dem klassischen WARP-Gehäuse reinpassen.