Tinkerforge/warp-charger #32: Erlaubter Ladestrom mqtt

fr-Fabix, 2023-04-14:
Hallo!

Ich habe die WARP1 mit 2.1.2 im Einsatz mit mqtt über homeassistant.

Ich versuche den Ladestrom zu setzen. Dabei gibt es folgendes Phänomen:

1. Ladestrom wird übernommen
2. Einige Sekunden später ist der Ladestrom 0

Siehe Video

https://user-images.githubusercontent.com/107420972/232071942-222ada05-1b98-4d68-8555-4f418a335187.mov

MattiasTF, 2023-04-17:
Kann ich leider nicht reproduzieren. Kannst du dich mal mit mosquitto_sub oder etwas ähnlichem auf das MQTT-Topic `warp/USH/evse/external_current` subscriben, um zu testen, ob die 0-Updates per MQTT kommen?

Hast du ansonsten zufällig EVCC oder sowas im Einsatz? EVCC setzt nämlich genau den gleichen Wert.

Falls das dein Problem noch nicht löst, poste mal ein Debug-Log.

fr-Fabix, 2023-04-17:
> Kann ich leider nicht reproduzieren. Kannst du dich mal mit mosquitto_sub oder etwas ähnlichem auf das MQTT-Topic `warp/USH/evse/external_current` subscriben, um zu testen, ob die 0-Updates per MQTT [...]

Du hast recht! Ich hatte testweise evcc laufen.... 

Danke!

closed 2023-04-17 by fr-Fabix

rediculum, 2025-05-22:
@fr-Fabix wie hast du es jetzt gelöst, dass du HA und evcc zusammen kombinieren konntest? die HA integration in warp2 setzt den Ladestrom ja über ```external_current```. Wäre es nicht besser, diesen über ```global_current``` zu setzen, denn bei mir funktioniert das nur so.

MattiasTF, 2025-05-22:
> Wäre es nicht besser, diesen über `global_current` zu setzen, denn bei mir funktioniert das nur so.

Der Wert für `global_current` wird im Flash-Speicher des EVSE gespeichert. Wenn du den automatisiert setzen lässt, ist der Flash in absehbarer Zeit hinüber. Davon rate ich ab.

`external_current` wird nur im RAM gehalten und ist für eine externe, automatisierte Steuerung vorgesehen, die den Wert beliebig oft ändern kann.

rediculum, 2025-05-22:
Wenn es um Überschussladen geht verstehe ich das, weil sich der Ladestrom dauernd ändern kann (Wolke zieht z.b. rüber), aber wenn ich "fast" laden will und nicht die vollen 16A möchte, setze ich das jetzt über ```global_current```, anders geht nicht. Aber auch bei PV Laden kommt es manchmal vor, dass ich nur bis max. 8A laden will, weil meine Huawei Batterie max 5KW Durchsatz hat und sonst vom Netz bezogen wird, was ja nicht im Interesse ist. Dann setze das auch über ```global_current```. Vielleicht mach ich da ein Anwendungsfehler?

MattiasTF, 2025-05-23:
In dem Fall ist `global_current` tatsächlich eine sinnvolle Möglichkeit. Wenn du das nur gelegentlich setzt und nicht mittels einer Regelung, die den Wert live anpasst, schadet das nicht. `external_current` wäre zwar prinzipiell besser, aber wenn HomeAssistant das schon an anderer setzt, kannst du das nicht überschreiben.