Tinkerforge/warp-charger #34: WEM-Anleitung: Schützanschluss besser beschreiben

MattiasTF, 2023-08-30:
Aktuell wird in der WEM-Anleitung zum Schütz für die Phasenumschaltung nur beschrieben, wie das Schütz an den WEM angeschlossen wird. Der Anschluss der Wallbox an das Schütz wird aber nicht erklärt.

andreas-bulling, 2023-11-12:
Und selbst diese "Beschreibung" ist nicht ausreichend. Es wird nicht erklärt, welche Anschlüsse an dem Rückmeldeteil (keine Ahnung wie das heißt) verwendet werden müssen. Ich habe das nur über einen Forumseintrag rausgefunden (die hinteren, 23 und 24 sind das meine ich).

batti, 2023-11-13:
Der Hinweis mit den Klemmen 23 + 24 steht nun in der Anleitung.

closed 2023-11-13 by batti

andreas-bulling, 2023-11-13:
Super, danke. Noch besser wäre ein kurzes Schaubild, wie man welche Anschlüsse miteinander verbinden muss. Also auch Relais, Inputs etc auf der EM-Seite