Tinkerforge/warp-charger #35: DC Version mit bidirektionalen Laden

janknieling, 2023-10-21:
Ich fände es sehr Sinnvoll wenn es eine Version des WARP Chargers gibt, welcher per DC laden kann aber auch bidirektionales Laden unterstützt.
Letzteres wäre meiner Meinung nach das wichtigste Feature, da man ja dadurch sein E Auto als Speicher nutzen kann, wo man überschüssigen Strom aus PV einspeichern und später wieder abrufen kann.
Am schönsten wäre es die Wallbox zu haben, die sowohl AC als auch DC Laden anbietet (und natürlich bidirektionales Laden). Ggf sogar eine Art Upgrade-Kit, mit dem man existierende WARP 2 Charger aufrüsten kann (z.B. über ein Erweiterungsmodul, was in einem separaten Kasten seitlich neben dem WARP 2 befestigt wird)

batti, 2023-10-23:
Hallo Jan,
das wäre in der Tat "schön". Technisch ist es aber ein sehr großer Unterschied ob eine Wallbox AC oder DC Laden anbietet (Wechselrichter). Von den Kosten her ist es dies auch. Wir verfolgen das Thema DC Laden, allerdings muss Jedem klar sein, dass eine DC Wallbox deutlich teurer ist, als eine AC Wallbox.

janknieling, 2023-10-23:
Ja das ist mir durchaus bewusst, dass das technisch komplexer ist. Es gibt ja leider keine einfache Möglichkeit den Gleichstrom aus der PV Anlage direkt zu einer Wallbox zu schicken.
Zum Thema Preis:
Ich denke, dass man je nachdem, was man mit der Wallbox vor hat den höheren Anschaffungspreis vernachlässigen kann.
Bei meinem Anwendungszweck wäre es jedenfalls so, da ich das ganze mit den Kosten für einen Batteriespeicher bei mir im Keller gegenrechne.
Bei so einem Speicher zahle ich viel mehr Geld pro kWh im Vergleich zur Anschaffung von einer Wallbox, die bidirektionales laden kann (ich hatte Angebote in Höhe von um die 8000€ für einen 8-10 kWh Speicher vorliegen, was ich sehr teuer finde, wenn man mal guckt, wie viel Batteriekapazität ein modernes E Auto hat. Da habe ich ja durchaus 40-80 kWh Batterien eingebaut).

closed 2025-07-13 by borg42