Tinkerforge/warp-charger #36: Add additional time periods in power history

andreas-bulling, 2023-11-15:
It'd be nice and useful to offer additional time periods in the power history in the web interface of the EnergyManager. Currently there are only two options - 48h and live. I think 24h and 1h would be useful / make more sense.

photron, 2023-11-27:
Instead of adding more fixed ranges we could make the 48h zoomable like this: 

https://leeoniya.github.io/uPlot/demos/zoom-ranger-grips.html

andreas-bulling, 2023-11-30:
Sure, fancy feature. But from my point of view not really necessary. I'd be happy to just select different time periods from a dropdown ;)

photron, 2024-04-17:
I added 24h and 12h views. 1h is not very usefull as the power history only records a new value every 4 minutes.

This will be part of the next WARP Energy Manager (2.1.0) and WARP Charger firmwares (2.3.1). No release date has been scheduled yet.

closed 2024-04-17 by photron