Tinkerforge/warp-charger #37: Power history doesn't show any data

andreas-bulling, 2023-11-15:
I am running firmware 1.0.91-65492465 on the EnergyManager and added a Fronius inverter using the new Sunspec protocol. The inverter got detected fine and is reporting values but these are not shown in the graph at the top.

[image]
[image]

MattiasTF, 2023-11-17:
This should be fixed in beta 5.

andreas-bulling, 2023-11-17:
will try and report back...

andreas-bulling, 2023-11-19:
works fine with beta 5

andreas-bulling, 2023-11-22:
stopped working again yesterday without having changed anything in the configuration

andreas-bulling, 2023-11-30:
Mine indeed does but there were also no values during the day for some reason. Now they are back...

closed 2023-11-30 by andreas-bulling