Tinkerforge/warp-charger #41: Add authentication to charger

andreas-bulling, 2024-01-27:
The web interface of the Energy Manager can be protected with a password. This provides some security and prevents users on the same network to make changes to the configuration and mess with the system.

The charger's web interface is unprotected at the moment, which is not good.

This (I think very important and urgent) feature request is to add authentication to the charger. I'd assume the code for this functionality could be copied more or less one-to-one from the EM.

rtrbt, 2024-01-29:
The charger already has this feature: You can set a password per user (or only for those users that are allows to log in) and then enable the login on the user management subpage.

closed 2024-01-29 by rtrbt

andreas-bulling, 2024-01-29:
Thanks for your reply. Just to make sure I understand correctly. I should
1) go to "User Management" and create a new user, e.g. "admin"
2) do what to prevent anyone else from logging into the web interface? Can the car still be charged without having to activate or "approve" anything before (through the web interface or a NFC card)?

rtrbt, 2024-01-30:
> go to "User Management" and create a new user, e.g. "admin"

Correct. And you have to enable the "Enable login" option.

> do what to prevent anyone else from logging into the web interface?

Only users that have a password set can log in. This is why you can only activate "Enable login" if at least one user has a password set.

> Can the car still be charged without having to activate or "approve" anything before (through the web interface or a NFC card)?

Yes. To require a NFC tag to charge, you would have to enable "Charge release" (directly below)

andreas-bulling, 2024-02-01:
Hm, but now it seems the EM cannot access the charger anymore...