Tinkerforge/warp-charger #67: WARP1 API: wrong /evse/automation_current?

Alestrix, 2025-01-25:
Hi,

I have an automation coming from my Shelly EM3 that limits the WARP's current if it detects overpower on one of the phases of the house's point of delivery.
When it triggers, one can see it in the WARP's GUI:
[image]

It can also be seen at slot 14 via the API:
``` 
$ curl --silent http://wallbox/evse/slots|jq '.[14]'
{
  "max_current": 8000,
  "active": true,
  "clear_on_disconnect": false
}
``` 

However, the `automation_current` is still reported to be 32 amps:
```
$ curl --silent http://wallbox/evse/automation_current|jq
{
  "current": 32000
}
```

So either `/evse/automation_current` is not updated or it refers to some other automation (according to the [reference](https://docs.warp-charger.com/docs/mqtt_http/api_reference/evse#evse_automation_current_warp1) it's "Der von der Automatisierung erlaubte Ladestrom.").

I'll work around it by using `slot[14]` (will that number be stable in future firmware versions?), but this still should be fixed.

WARP 1, Version 2.6.6+675aea10 (also tested with 2.6.1 showing the same behavior)

rtrbt, 2025-01-27:
> So either /evse/automation_current is not updated

Exactly that. Should be fixed with this firmware (you have to extract the zip first, Github does not allow to upload bin files):

[warp_firmware-NIGHTLY_2_6_6_679748cd_c4005e8a6f81e55__merged.bin.zip](https://github.com/user-attachments/files/18555740/warp_firmware-NIGHTLY_2_6_6_679748cd_c4005e8a6f81e55__merged.bin.zip)

Thanks for the bug report!

Alestrix, 2025-02-01:
Working, thanks!
```
$ curl --silent http://wallbox/evse/automation_current|jq
{
  "current": 32000
}
$ curl http://wallbox/automation_trigger/overpower_true
$ curl --silent http://wallbox/evse/automation_current|jq
{
  "current": 8000
}
```
👍

closed 2025-02-01 by Alestrix