Tinkerforge/warp-charger #78: Feature Request: Minimalüberschuss für PV Laden manuell einstellen

Alestrix, 2025-04-26:
Moin!

Ich habe eine WARP1 mit externen Zählern, die per API ihre Werte an die Box melden. So gut wie alles funktioniert perfekt - bis auf den korrekten Start beim PV-Überschussladen.
Bei einphasigem Laden (z.B. haben auch viele Hybrid-Fahrzeuge nur einen 1-phasigen OBC) könnte das PV-Laden schon bei 1380W starten, die Box weiß aber natürlich nicht, dass beim Starten nur einphasig geladen werden würde und wartet, bis genug Überschuss für 3-phasiges Laden vorhanden ist, was unter Umständen den ganzen Tag nicht passiert.

Daher mein Feature Request: Eine weitere Einstellung im PV-Überschussladen Menü, wo man (neben der Min+PV Mindestladeleistung, die man schon auf 1380W stellen kann) auch eine "Ladestart-Untergrenze" selbst definieren kann. Falls das zu aufwendig ist, wäre vielleicht eine Einstellung, die man per API vornehmen kann, einfacher umzusetzen?

Danke im Voraus!
Alex

MattiasTF, 2025-04-30:
Wenn ich mich recht erinnere, hattest du dir dieses Feature schon in unserem Forum gewünscht. Leider stehen die Chancen dafür relativ schlecht, da das ein Sonderfeature für WARP1 wäre. Im Gegensatz zu dem, was bei vielen anderen Herstellern zu erwarten wäre, bekommen die alten WARP1 bei uns immer noch neue Features, da wir durch unser modules System neue Features für die neueren WARP-Generationen oft ohne Aufwand auch für WARP1 verfügbar machen können. Da alle unsere Todo-Listen mehr als voll sind, halte ich es aber für unwahrscheinlich, dass es noch neue Features speziell für WARP1 geben wird, sofern sie nicht von der Community beigesteuert werden.

Alestrix, 2025-06-29:
Hi

erst mal sorry für die späte Rückmeldung!

> da das ein Sonderfeature für WARP1 wäre

Ich dachte, dass das eigentlich nicht nur für die WRAP1 ein Thema wäre, aber möglicherweise missverstehe ich auch die Funktionsweise der anderen beiden Boxen. 

Wenn z.B. eine WARP2 dreiphasig angeschlossen ist, aber ein Fahrzeug PV-geladen werden soll, dass nur einphasig laden kann - woher weiß die Box, dass schon bei 1380W Überschuss gestartet werden kann? 

Bei WARP3 stellt man wahrscheinlich in der GUI einfach auf eine Phase um, aber die WARP2 kann das ja noch nicht.

Grüße
Alex

closed 2025-07-13 by borg42